platformio/platformio-vscode-ide #1218: Installation Manager

AlessandroAU, 2019-10-20:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 6.1.7601, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Alessandro\.platformio\.cache\tmp-125087fjOfzWyWkZt\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\Alessandro\.platformio\.cache\tmp-125087fjOfzWyWkZt\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\Alessandro\.platformio\.cache\tmp-125087fjOfzWyWkZt\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\Alessandro\AppData\Local\Programs\Python\Python37-32\Lib\ntpath.py", line 526, in abspath
    return _getfullpathname(path)
TypeError: _getfullpathname: path should be string, bytes or os.PathLike, not NoneType

Please use the *system* python to run this script

	at t.value (c:\Users\Alessandro\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

ivankravets, 2019-10-20:
Do you have antivirus tools in a system?

AlessandroAU, 2019-10-21:
No antivirus, seems like a permission problem though. I had to delete the entire .platformio folder and do a fresh install

closed 2019-10-21 by ivankravets