platformio/platformio-vscode-ide #1219: Installation Manager

robertassarva, 2019-10-20:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\robertas.sarva\.platformio\.cache\tmp-18652Rd5uMPu160Y6\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\robertas.sarva\.platformio\.cache\tmp-18652Rd5uMPu160Y6\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\robertas.sarva\.platformio\.cache\tmp-18652Rd5uMPu160Y6\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\robertas.sarva\.platformio\.cache\tmp-18652Rd5uMPu160Y6\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\robertas.sarva\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\robertas.sarva\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\robertas.sarva\\.platformio\\python37'
New python executable in C:\Users\robertas.sarva\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\robertas.sarva\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

ivankravets, 2019-10-20:
Do you use Antivirus tools in a system?

robertassarva, 2019-10-20:
No, only standard which came with windows.
I removed .platformio folder. Got an error about .py file in use.
Killed python  Reinstalled platformio  whith in VS CODE.
Redownloaded Python 3.8, and reregistered in PATH.
And now it works.

Robertas Šarva
IT ir verslo procesų analitikas

Vildika UAB
P. Lukšio g. 18, Vilnius
+370 655 05845
robertas.sarva@vildika.lt

www.vildika.ltwww.kedziucentras.lt [image: Facebook icon]
<https://www.facebook.com/vildika/>   [image: Instagram icon]
<https://www.instagram.com/vildika.uab/>

[image: Banner] <https://www.vildika.lt/>

ivankravets, 2019-10-20:
See solution here https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-20 by ivankravets