platformio/platformio-vscode-ide #1221: Installation Manager

kristapsdravnieks, 2019-10-20:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\krist\.platformio\.cache\tmp-2236hV4EEu7LJOEI\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\krist\.platformio\.cache\tmp-2236hV4EEu7LJOEI\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\krist\.platformio\.cache\tmp-2236hV4EEu7LJOEI\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\krist\.platformio\.cache\tmp-2236hV4EEu7LJOEI\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\krist\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\krist\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\krist\\.platformio\\python37'
New python executable in C:\Users\krist\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\krist\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

ivankravets, 2019-10-20:
Hi Kristaps, 

Could you provide more details on this issue? We a lot of similar issues and don't understand where is the root of problem.

Do you have anitivrus tools in system?

ivankravets, 2019-10-20:
See solution here https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-20 by ivankravets