platformio/platformio-vscode-ide #1228: Installation Manager

JackFrost80, 2019-10-20:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: 

	at t.value (c:\Users\JackFrost\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

neilh10, 2019-10-21:
I have the same. Two days ago I completely removed VSC from global. After reinstalling VSC as local,   PlatformIO started automatically. I was able to build and download.
Today on starting up I get this problem.
I retried with Admin Priv - and no difference

When trying to follow 
http://docs.platformio.org/en/latest/installation.html#virtual-environment

I get 
python -c "import os; print(os.path.join(os.getenv('PLATFORMIO_CORE_DIR', os.path.join(os.path.expanduser('~'), '.platformio')), 'penv'))"
No pyvenv.cfg file
When executing from a cmd line prompt
C:\Users\neilh77>python --version
Python 3.7.2
( that is installed at c:\python37) 
There was a Python3.4 but I removed it through Control Panel-Programs and Features
However c:\Pythong34 was still there and PIO was still picking up on it in $env:Path - so manually deleted it,
However $env:Path is still showing it
C:\Users\neilh77\.platformio\penv\Scripts;C:\Users\neilh77\.platformio\penv;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Python37\Scripts\;C:\Python37\;C:\Python34\;C:\Python34\Scripts;
....

%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17134, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\neilh77\.platformio\.cache\tmp-8248hr0YrFu14LoH\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\neilh77\.platformio\.cache\tmp-8248hr0YrFu14LoH\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\neilh77\.platformio\.cache\tmp-8248hr0YrFu14LoH\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\neilh77\.platformio\.cache\tmp-8248hr0YrFu14LoH\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\neilh77\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Python37'
New python executable in C:\Users\neilh77\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\neilh77\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

neilh10, 2019-10-21:
An update -following a hint in http://docs.platformio.org/en/latest/installation.html#virtual-environment 
I found I had an earlier c:\python34 which I removed through  Control Panel-->Programs and Features and then deleted off disk c:\python34  (leaving c:\python37)
Then I found I had a  C:\Users\neilh77\.platformio\python27  - which I deleted .
Then I rebooted,  started up VSC/PlatformiIO, it installed updates successfully,  and can now compiles OK. Hurrah!! 
Really appreciate the work that goes into making PlatformIO an amazing development IDE.!!!

For my PIO Terminal that invokes PowerShell, when I do $Env:path it still lists the C:\Python34  - even though its now gone:
$Env:path
C:\Users\neilh77\.platformio\penv\Scripts;C:\Users\neilh77\.platformio\penv;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Python37\Scripts\;C:\Python37\;C:\Python34\;C:\Python34\Scripts;
....

ivankravets, 2019-10-21:
We don't support Python 3.4, please remove it.

As for the your issue, see solution https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-21 by ivankravets