platformio/platformio-vscode-ide #1237: Installation Manager

ejngnng, 2019-10-21:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17134, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at t.value (c:\Users\ningjiang\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

ivankravets, 2019-10-21:
Could you provide installer log from `c:\Users\aukev\.platformio\.cache` dir? What is your Python version?
```
python --version
```

ivankravets, 2019-11-16:
Please reopen if you still need help

closed 2019-11-16 by ivankravets

ejngnng, 2020-01-15:
reinstall platformio solve the problem， thanks @ivankravets