platformio/platformio-vscode-ide #1240: Installation Manager

ndtcenter, 2019-10-21:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\yujin\.platformio\.cache\tmp-6848RxDRcyltK7zw\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\yujin\.platformio\.cache\tmp-6848RxDRcyltK7zw\virtualenv-16.7.5\virtualenv.py", line 860, in main
    create_environment(
  File "C:\Users\yujin\.platformio\.cache\tmp-6848RxDRcyltK7zw\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\yujin\.platformio\.cache\tmp-6848RxDRcyltK7zw\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "D:\pf\Python\Python38\lib\shutil.py", line 259, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\yujin\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'D:\\pf\\Python\\Python38'
New python executable in C:\Users\yujin\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\yujin\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

langph, 2019-10-21:
Try restarting your computer, worked for me.

ivankravets, 2019-10-21:
See solution https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-21 by ivankravets