platformio/platformio-vscode-ide #1244: Installation Manager

mhollow56, 2019-10-21:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\mikeh\.platformio\.cache\tmp-16408dCoLG8yutL59\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\mikeh\.platformio\.cache\tmp-16408dCoLG8yutL59\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\mikeh\.platformio\.cache\tmp-16408dCoLG8yutL59\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\mikeh\.platformio\.cache\tmp-16408dCoLG8yutL59\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\mikeh\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\mikeh\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\mikeh\\.platformio\\python37'
New python executable in C:\Users\mikeh\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\mikeh\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

pekkalehtikoski, 2019-10-22:
I believe this is mostly a known minor issue "permission denied #1213" in PlatformIO 1.9.2 windows installation. In addition, make sure that you have Python 3.8 installed and used (easiest way to make sure it is used on Windows is uninstall older Python versions). When installing python for windows, I select custom installation to install it for all users and to register environment variables.

There are many ways to work around this, but simplest:

Once this error appears, close visual studio code, reboot windows and start visual studio code again. Installation should resume itself automatically.
Or, uninstall Platform IO, reboot windows computer, install Platform IO.

ivankravets, 2019-10-22:
See solution here https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-22 by ivankravets