platformio/platformio-vscode-ide #1247: Can not install extension

charclo, 2019-10-22:
After installing the platformio extension, the "PlatformIO Home” button on the bottom PlatformIO Toolbar isn't shown.
When I try to add it with the Command Palette I get this error: command 'platformio-ide.showHome' not found

ivankravets, 2019-10-22:
Could you provide errors from VSCode > Help > Toggle developer tools > Console?

charclo, 2019-10-22:
[image]

charclo, 2019-10-22:
I installed it successfully on my Mac.
There I did see: Installing PlatformIO Core...
Please do not close this window and do not open other folders until this process is completed.
PlatformIO IDE installed successfully.

Please restart VSCode.

I never got this message on my windows 10 computer

ivankravets, 2019-11-16:
Did you fix it?

charclo, 2019-11-16:
No, it still doesn’t work on my windows 10 computer. Can I provide more information?

ivankravets, 2019-11-16:
Could you provide AnyDesk session to debug this issue? Please run https://anydesk.com/en and send ID to me@ikravets.com. Thanks!

closed 2019-11-21 by ivankravets