platformio/platformio-vscode-ide #1250: Installation Manager

dineshvyas, 2019-10-22:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x047994F0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x04799450>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x047995F0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x04766970>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x04766E10>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/platformio/
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio

	at o.runCommand (c:\Users\Dinesh Vyas\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:33258)
	at ChildProcess.h (c:\Users\Dinesh Vyas\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\Dinesh Vyas\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

dineshvyas, 2019-10-22:
windows defender disabled but problem persist.

ivankravets, 2019-10-22:
>  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x047994F0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': 

Try manually to uninstall system Pythons and install the latest Python http://docs.platformio.org/en/latest/faq.html#install-python-interpreter

closed 2019-10-22 by ivankravets