platformio/platformio-vscode-ide #1251: Start PIO Home Server

TenoTrash, 2019-10-22:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Linux, 4.15.0-66-generic, x64

%23 Exception
```
Error: Error: Could not install 'contrib-piohome' with version requirements '^2.3.2' for your system 'linux_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at /home/teno/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:17820
	at ChildProcess.h (/home/teno/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

piotrbrytan, 2019-10-22:
%23 Description of problem
Same issue here

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Linux, 5.0.0-31-generic, x64

%23 Exception
```
Error: Error: Could not install 'contrib-piohome' with version requirements '^2.3.2' for your system 'linux_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at /home/piotr/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:17820
	at ChildProcess.h (/home/piotr/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

Aurawin, 2019-10-22:
Same problem...

atbrunner@Inspiron:~$ pio homeCorePackageManager: Installing contrib-piohome @ ^2.3.2
Warning! Package Mirror: The 'sha1' sum '17c2a8e8c9c426660f85d5132ac31da752518f49' of downloaded file 'contrib-piohome-2.3.3.tar.gz' is not equal to remote '3cc39a68453fdb6420e1c07766250c8a0d92b3f6'
Looking for another mirror...
Downloading  ... etc...
   
Warning! Package Mirror: The 'sha1' sum '17c2a8e8c9c426660f85d5132ac31da752518f49' of downloaded file 'contrib-piohome-2.3.3.tar.gz' is not equal to remote '3cc39a68453fdb6420e1c07766250c8a0d92b3f6'

Looking for another mirror...
Error: Could not install 'contrib-piohome' with version requirements '^2.3.2' for your system 'linux_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

sirdel, 2019-10-23:
VScode: 1.39.2
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Linux x64 4.19.0-6-amd64
PlatformIO: 4.0.3

Same problem here.

CorePackageManager: Installing contrib-piohome @ ^2.3.2
Downloading  [####################################]  100%          
Warning! Package Mirror: The 'sha1' sum '17c2a8e8c9c426660f85d5132ac31da752518f49' of downloaded file 'contrib-piohome-2.3.3.tar.gz' is not equal to remote '3cc39a68453fdb6420e1c07766250c8a0d92b3f6'
Looking for another mirror...

The solution at http://bit.ly/faq-package-manager worked by manually downloading the zip into the core_dirs folder, but the next time I imported an example project, the package had been removed !!

The python sha1 results :-

python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
(sys.version_info(major=2, minor=7, micro=16, releaselevel='final', serial=0), <sha1 HASH object @ 0x7f5e4f9d7698>)
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
sys.version_info(major=3, minor=7, micro=3, releaselevel='final', serial=0) <sha1 HASH object @ 0x7f48feeeb710>

ivankravets, 2019-10-23:
Could you try to restart VSCode? I've just redeployed a package to CDN.

ivankravets, 2019-10-23:
Sorry for the issue. Just restart VSCode. Everything should work now.

closed 2019-10-23 by ivankravets

TenoTrash, 2019-10-23:
Yes, works like a charm! Thank you!