platformio/platformio-vscode-ide #1253: Can't install on mac

kammoh, 2019-10-22:
```
Error: Conda Virtualenv: 
[Errno 13] Permission denied: '/.cph_tmpu5d12d22'
[Errno 13] Permission denied: '/.cph_tmpf2vv3fu4'
[Errno 13] Permission denied: '/.cph_tmpoyggl5xh'
	at o.runCommand (/Users/kamyar/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:30256)
	at ChildProcess.h (/Users/kamyar/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

Things I tried and that did not help:
- Delete ~/.vscode and ~/.platformio
- manually create python virtualenv

Vscode Version: 1.39.2
Commit: 6ab598523be7a800d7f3eb4d92d7ab9a66069390
OS: Darwin x64 19.0.0 (macOS 10.15)
Python: Homebrew python3.7.4

kammoh, 2019-10-22:
Now getting this: :/
```
Error: Error: Could not install 'contrib-piohome' with version requirements '^2.3.2' for your system 'darwin_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at /Users/kamyar/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:17820
	at ChildProcess.h (/Users/kamyar/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

kammoh, 2019-10-22:
Recommendations on #1108 does not help

ivankravets, 2019-10-23:
Just restart VSCode, everything should work. Sorry for the issue.

closed 2019-10-23 by ivankravets