platformio/platformio-vscode-ide #1259: platformio keeps reinstalling

HamzaHajeir, 2019-10-22:
Hi

I'm facing an issue whenever there's a new version of PIO xD

I couldn't fix it this time.

PIO keeps re-installing penv/Scripts in the startup of  VS Code [latest version].

Giving me this message : 

> Installing PlatformIO Core...
> Please do not close this window and do not open other folders until this process is completed.
> PlatformIO IDE installed successfully.
> 
> Please restart VSCode.

I've tried : 
a- Uninstall / Install platformIO from VSC extensions.
b- install manually by creating [virtualenv ](http://docs.platformio.org/en/latest/installation.html#creating). 
c- Uninstalled all python interpreters in my machine (Windows 10). 
d- Installed PIO again manually from VSCode, No success.

In b : after I install PIO successfully. by running VSCode, reinstall happens again :\ 

Hope to find a solution to this issue soon.

ivankravets, 2019-10-23:
Simple solution:

1. Uninstall all Pythons from system via system program manager
2. Install the latest Python 3.8 http://docs.platformio.org/en/latest/faq.html#install-python-interpreter
3. Remove `%USERPROFEIL%/.platformio` folder
4. Start VSCode.

Did it help?

HamzaHajeir, 2019-10-23:
Thank you @ivankravets, 

It Worked well (y)

closed 2019-10-23 by HamzaHajeir