platformio/platformio-vscode-ide #1262: Start PIO Home Server

kruzhkov, 2019-10-22:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Darwin, 18.6.0, x64

%23 Exception
```
Error: Error: Could not install 'contrib-piohome' with version requirements '^2.3.2' for your system 'darwin_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at /Users/sergeykruzhkov/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:17820
	at ChildProcess.h (/Users/sergeykruzhkov/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

driscollta, 2019-10-23:
Similar problem new installation of PlatformIO for VSCode

Error: Error: Could not install 'contrib-piohome' with version requirements '^2.3.2' for your system 'linux_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at /home/tommy_d/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:17820
	at ChildProcess.h (/home/tommy_d/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

ivankravets, 2019-10-23:
Could you open PlatformIO IDE Terminal and type 
```
pio home
```

You will see a more detailed error. It looks like your Python installation has a problem with TLS connection where we can not get packages list.

ivankravets, 2019-10-23:
Sorry for the issue. Just restart VSCode. Everything should work now.

closed 2019-10-23 by ivankravets