platformio/platformio-vscode-ide #1263: Start PIO Home Server

snailspeed, 2019-10-23:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.37.190418' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at c:\Users\123\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:17820
	at ChildProcess.h (c:\Users\123\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.EventEmitter.emit (domain.js:442:20)
	at ChildProcess.cp.emit (c:\Users\123\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-23:
1. Please close VSCode
2. Open System Task manager and kill all `python.exe` processes
3. Start VSCode.

Did it help?

ivankravets, 2019-11-16:
Please reopen if you still need help

closed 2019-11-16 by ivankravets

oxihormesis, 2020-06-08:
this problem persists in 1.10.0 on linux_x86

ivankravets, 2020-06-08:
@oxihormesis could you reproduce this issue if open PlaformIO IDE Terminal (terminal icon on the bottom toolbar) and type `pio home`?

oxihormesis, 2020-06-08:
@ivankravets I just saw in another issue where you recommended to install beta 2.0- did this and it's installed without errors!

ivankravets, 2020-06-08:
Great! Please monitor that topic in our forum, we plan to release Beta 2 soon. It will bring something sweet 🍬  :)

oxihormesis, 2020-06-08:
👀 Awesome! I'll watch keenly for that