platformio/platformio-vscode-ide #1265: Start PIO Home Server

buhadram, 2019-10-23:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Linux, 5.3.7, x64

%23 Exception
```
Error: Error: Could not install 'contrib-piohome' with version requirements '^2.3.2' for your system 'linux_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at /home/lutfi/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:17820
	at ChildProcess.h (/home/lutfi/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

0n3-70uch, 2019-10-23:
+1 confirm
Same problem with Atom-IDE

netbomo, 2019-10-23:
+1 same with 
VSCode 1.39.2
PIO IDE : 1.9.2
System : debian buster, 4.19.0-6-amd64

Fetzer54, 2019-10-23:
+1 same with
VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 6.1.7601, x64

michelpa, 2019-10-23:
Same problem
Ubuntu 19.04

ivankravets, 2019-10-23:
Could you try to restart VSCode. Does it work now?

netbomo, 2019-10-23:
Yes thanks!

ivankravets, 2019-10-23:
Sorry for the issue!

closed 2019-10-23 by ivankravets

0n3-70uch, 2019-10-23:
I tried these steps:
- restart 
- deleting .platformio
- restart

Always crash...

ivankravets, 2019-10-23:
@0n3-70uch  see solution https://github.com/platformio/platformio-vscode-ide/issues/1259#issuecomment-545377055

0n3-70uch, 2019-10-23:
> @0n3-70uch see solution [#1259 (comment)](https://github.com/platformio/platformio-vscode-ide/issues/1259#issuecomment-545377055)

Works fine now.... thank you.