platformio/platformio-vscode-ide #1269: Start PIO Home Server

GilsonMendes, 2019-10-23:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Darwin, 16.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Users/gilsonmendes/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:17820
	at ChildProcess.h (/Users/gilsonmendes/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-24:
Could you open PlatformIO IDE Terminal, see icon on the bottom toolbar and run:
```
python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
```

Do you see the output?

szymoni, 2019-10-27:
@ivankravets, I have just run the request: 
raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='api.platformio.org', port=443): Max retries exceeded with url: /stats (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1076)')))

ivankravets, 2019-10-28:
Please install the latest Python 3 from https://www.python.org/

Open system terminal and type
```
rm -rf ~/.platformio/penv
```

Restart VSCode.

closed 2019-10-28 by ivankravets