platformio/platformio-vscode-ide #1281: Installation Manager

samohill, 2019-10-24:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\pan_c\.platformio\.cache\tmp-17200Xb0m0BZ4Wy2p\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\pan_c\.platformio\.cache\tmp-17200Xb0m0BZ4Wy2p\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\pan_c\.platformio\.cache\tmp-17200Xb0m0BZ4Wy2p\virtualenv-16.7.5\virtualenv.py", line 1173, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\pan_c\.platformio\.cache\tmp-17200Xb0m0BZ4Wy2p\virtualenv-16.7.5\virtualenv.py", line 1019, in install_wheel
    _install_wheel_with_search_dir(download, project_names, py_executable, search_dirs)
  File "C:\Users\pan_c\.platformio\.cache\tmp-17200Xb0m0BZ4Wy2p\virtualenv-16.7.5\virtualenv.py", line 1046, in _install_wheel_with_search_dir
    config = _pip_config(py_executable, python_path)
  File "C:\Users\pan_c\.platformio\.cache\tmp-17200Xb0m0BZ4Wy2p\virtualenv-16.7.5\virtualenv.py", line 1124, in _pip_config
    raise_on_return_code=False,
  File "C:\Users\pan_c\.platformio\.cache\tmp-17200Xb0m0BZ4Wy2p\virtualenv-16.7.5\virtualenv.py", line 921, in call_subprocess
    env=env,
  File "C:\Users\pan_c\AppData\Local\Programs\Python\Python36\Lib\subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "C:\Users\pan_c\AppData\Local\Programs\Python\Python36\Lib\subprocess.py", line 997, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Acceso denegado

Please use the *system* python to run this script
Error [WinError 5] Acceso denegado while executing command C:\Users\pan_c\.platformio\penv\Scripts -m pip config list

	at t.value (c:\Users\pan_c\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

nikil511, 2019-10-27:
Solution is described in [#1213](https://github.com/platformio/platformio-vscode-ide/issues/1213)

nikil511, 2019-10-27:
Solution described in #1213

ivankravets, 2019-11-16:
See solution https://github.com/platformio/platformio-vscode-ide/issues/154

closed 2019-11-16 by ivankravets