platformio/platformio-vscode-ide #1286: Can't install platformio-ide (in neither Atom nor VSCode)

xhain, 2019-10-26:
`Error: Conda Virtualenv: [Errno 13] Permission denied: '/.cph_tmpowhgqu2k' [Errno 13] Permission denied: '/.cph_tmp7d6emnsu' [Errno 13] Permission denied: '/.cph_tmpscqxi8ec' [Errno 13] Permission denied: '/.cph_tmpk86aw4lq' [Errno 13] Permission denied: '/.cph_tmpewoqimnq' [Errno 13] Permission denied: '/.cph_tmping612qz' [Errno 13] Permission denied: '/.cph_tmp7hzq9qbw' [Errno 13] Permission denied: '/.cph_tmppyv_xwo5' at o.runCommand (/Users/max/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:30256) at ChildProcess.h (/Users/max/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292) at ChildProcess.emit (events.js:182:13) at maybeClose (internal/child_process.js:962:16) at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)`

OSX 10.13.6 (17G65)

ivankravets, 2019-10-28:
> Conda Virtualenv:

Please upgrade Conda installation or remove it and use the official Python 3.

closed 2019-10-28 by ivankravets

xhain, 2019-10-28:
Thank you @ivankravets !
I'm not proud of it, but you were right: Merely updating `conda` did the job for me 🙈