platformio/platformio-vscode-ide #1289: Start PIO Home Server

rdfwxff3, 2019-10-27:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 6.3.9600, ia32

%23 Exception
```
Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.27.190418' for your system 'windows_x86'.

Please try this solution -> http://bit.ly/faq-package-manager

	at c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:17820
	at ChildProcess.h (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-11-16:
Do you use antivirus tools? They can block PlatformIO Installer.

Try to open PlatformIO IDE Terminal and type
```
pio home
```

Do you see any errors?

rdfwxff3, 2019-11-17:
已解决谢谢.

closed 2019-11-18 by ivankravets