platformio/platformio-vscode-ide #1291: Start PIO Home Server

yulong88888, 2019-10-27:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.18363, x64

%23 Exception
```
Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.38.190418' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at c:\Users\admin\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:17820
	at ChildProcess.h (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

yulong88888, 2019-10-27:
I reinstalled the windows system, and also reinstalled the PIO. I have not encountered any problems before, but this time it is unfortunate that PIO Home can't open. I really like this tool, thanks for providing such excellent tools.

szymoni, 2019-10-27:
The very same problem with mine installation, just after latest PIO update:

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17134, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

yulong88888, 2019-10-28:
My problem is solved, probably because of the network, I use my mobile hotspot just fine.

closed 2019-10-28 by yulong88888

cgrrty, 2019-10-30:
Too bad, I also encountered the same problem, I am working on it.