platformio/platformio-vscode-ide #1303: Start PIO Home Server

IcebergXTY, 2019-10-28:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.36.190418' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at c:\Users\Iceberg\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:17820
	at ChildProcess.h (c:\Users\Iceberg\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\Iceberg\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-11-16:
Do you use antivirus tools? They can block PlatformIO Installer.

Try to open PlatformIO IDE Terminal and type
```
pio home
```

Do you see any errors?

IcebergXTY, 2019-11-17:
Yeah, I use Tencent PC Manager and the command result is below.
emmm... I live in Chinese mainland, is it possible the GFW block it?
---
PS C:\Users\Iceberg> pio home

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

Open PIO Home in your browser by this URL => http://127.0.0.1:8008
PIO Home has been started. Press Ctrl+C to shutdown.

ivankravets, 2019-11-18:
Does it work if you open now http://127.0.0.1:8008 in browser?

IcebergXTY, 2019-11-19:
Yeah, It works well now.

closed 2019-11-19 by ivankravets