platformio/platformio-vscode-ide #1307: Start PIO Home Server

nimrod86, 2019-10-28:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Error: Traceback (most recent call last):
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\__main__.py", line 90, in main
    cli(None, None, None)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\click\core.py", line 1134, in invoke
    Command.invoke(self, ctx)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\__main__.py", line 36, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 41, in on_platformio_start
    after_upgrade(ctx)
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 134, in after_upgrade
    app.clean_cache()
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\app.py", line 309, in clean_cache
    with ContentCache() as cc:
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\app.py", line 187, in __enter__
    self.delete()
  File "c:\users\cilli\.platformio\penv\lib\site-packages\platformio\app.py", line 296, in delete
    with open(self._db_path, "w") as fp:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\cilli\\.platformio\\.cache\\db.data'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\cilli\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:17820
	at ChildProcess.h (c:\Users\cilli\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\cilli\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-30:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1213

ivankravets, 2019-10-30:
Also, try to remove `C:\\Users\\cilli\\.platformio\\.cache` folder and restart VSCode.

closed 2019-10-30 by ivankravets