platformio/platformio-vscode-ide #1321: Installation Manager

hassanmj, 2019-10-31:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.3
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Hassan.Jajai\.platformio\.cache\tmp-10664amiC1dCM5ZqT\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\Hassan.Jajai\.platformio\.cache\tmp-10664amiC1dCM5ZqT\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\Hassan.Jajai\.platformio\.cache\tmp-10664amiC1dCM5ZqT\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\Hassan.Jajai\.platformio\.cache\tmp-10664amiC1dCM5ZqT\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\Hassan.Jajai\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\Hassan.Jajai\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\Hassan.Jajai\\AppData\\Local\\Programs\\Python\\Python37-32'
New python executable in C:\Users\Hassan.Jajai\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\Hassan.Jajai\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:32319)
```

1technophile, 2019-11-01:
I'm having the same issue after having launch the update today

ivankravets, 2019-11-16:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1213. See solution https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-11-16 by ivankravets