platformio/platformio-vscode-ide #1322: Installation Manager

DenisMtfl, 2019-11-01:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.3
System: Windows_NT, 10.0.18363, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x000030e4 (most recent call first):

    at t.value (c:\Users\Denis\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\installer\stages\platformio-core.js:303:14)
```

thb584Git, 2019-11-09:
Same problem here, any feedback, workaround or solution yet? 
Problem occurs after trying a complete reinstallion hoping to solve a problem with programmer for AVR devices not working with PlatformIO IDE.

ivankravets, 2019-11-16:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

thb584Git, 2019-11-16:
Here you go, attached requested output plus programming log w/wo custom in pio.ini

Afterwards Upgraded pip version 18.1 to version 19.3.1 as recommended with same result.

 

Best Regards

Thomas

 

Host-M win7/64bit:

C:\Users\thomas\Documents\PlatformIO\Projects\GitRepos>echo %PATH%
C:\Users\thomas\.platformio\penv\Scripts;C:\Users\thomas\.platformio\penv;C:\progs\WinAVR-20100110\bin;C:\progs\WinAVR-20100110\utils\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft VS Code\bin;C:\Users\thomas\AppData\Local\Microsoft\WindowsApps;;C:\progs\WinAVR-20100110\bin;C:\progs\WinAVR-20100110\utils\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft VS Code\bin;C:\Users\thomas\AppData\Local\Microsoft\WindowsApps;

C:\Users\thomas\Documents\PlatformIO\Projects\GitRepos>python --version
Python 2.7.13

C:\Users\thomas\Documents\PlatformIO\Projects\GitRepos>pip --version
pip 18.1 from c:\users\thomas\.platformio\penv\lib\site-packages\pip (python 2.7)

C:\Users\thomas\Documents\PlatformIO\Projects\GitRepos>pip list
Package          Version
---------------- ---------
bottle           0.12.17
certifi          2019.9.11
chardet          3.0.4
Click            7.0
colorama         0.4.1
idna             2.8
marshmallow      2.20.5
pip              18.1
platformio       4.1.0
pyelftools       0.25
pyserial         3.4
requests         2.22.0
semantic-version 2.8.2
setuptools       41.4.0
tabulate         0.8.5
urllib3          1.25.6
wheel            0.33.6
You are using pip version 18.1, however version 19.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

Custom set:

Processing myenv (platform: atmelavr; framework: arduino; board: uno)
------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR 1.15.0 > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.2, tool-avrdude 1.60300.190424 (6.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for FastLed library in registry
Found: https://platformio.org/lib/show/126/FastLED
LibraryManager: Installing id=126
FastLED @ 3.3.2 is already installed
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MobaLedLib> 0.9.2
|   |-- <FastLED> 3.3.2
|   |   |-- <SPI> 1.0
|   |   |-- <SoftwareSerial> 1.0
[... 102 lines cut ...]
.pio\build\myenv\libea2\FastLED_ID126\power_mgt.cpp.o .pio\build\myenv\libea2\FastLED_ID126\wiring.cpp.o .pio\build\myenv\src\RGB-LED8.cpp.o -L.pio\build\myenv -Wl,--start-group .pio\build\myenv\libc25\libSoftwareSerial.a .pio\build\myenv\libd86\libSPI.a .pio\build\myenv\lib288\libMobaLedLib_ID5987.a .pio\build\myenv\libFrameworkArduinoVariant.a .pio\build\myenv\libFrameworkArduino.a -lm -Wl,--end-group
MethodWrapper(["checkprogsize"], [".pio\build\myenv\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=         ]   8.6% (used 176 bytes from 2048 bytes)
PROGRAM: [==        ]  15.8% (used 5086 bytes from 32256 bytes)
.pio\build\myenv\firmware.elf  :
section                    size      addr
.data                        38   8388864
.text                      5048         0
.bss                        138   8388902
.comment                     48         0
.note.gnu.avr.deviceinfo     64         0
.debug_info                1524         0
.debug_abbrev              1442         0
.debug_line                  26         0
.debug_str                  520         0
Total                      8848
<lambda>(["program"], [".pio\build\myenv\firmware.hex"])
AVAILABLE: stk500v2
CURRENT: upload_protocol = stk500v2
MethodWrapper(["program"], [".pio\build\myenv\firmware.hex"])
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [program] Explicit exit, status 1
==================================================== [FAILED] Took 2.68 seconds ====================================================

ivankravets, 2019-11-16:
Try to uninstall all Pythons from your machine and install the latest http://docs.platformio.org/en/latest/faq.html#install-python-interpreter

closed 2019-11-16 by ivankravets