platformio/platformio-vscode-ide #1323: Cannot activate extension 'PlatformIO IDE' because it depends on extension 'ms-vscode.cpptools', which failed to activate.

UPUPGOO, 2019-11-02:
oh,there is no home 🏠 icon on the bottom status bar or PlatformIO Activity Bar.
I opened vscode->help->toggle developer tools->console, I got some error.

> Cannot activate extension 'PlatformIO IDE' because it depends on extension 'ms-vscode.cpptools', which failed to activate.

So, what should I do?

UPUPGOO, 2019-11-02:
I removed ./vscode/extensions/ms-vscode.cpptools and uninstalled the extension C/C++.
The I reinstalled platformIO and everything was ok.

closed 2019-11-02 by UPUPGOO

MagTun, 2021-08-03:
The error I got was:
Command 'PlatformIO: Home' resulted in an error (command 'platformio-ide.showHome' not found)

The path is C:\Users\user\.vscode\extensions.   

I had to uninstall other extension before being able to uninstall C/C++ (Arduino). If you don't have the C/C++ extension install when you install PlatformIO you need to restart VScode after both are installed.