platformio/platformio-vscode-ide #1326: Start PIO Home Server - python version and solution

voltane, 2019-11-02:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.3
System: Linux, 4.4.0-165-generic, x64
**OS**: Ubuntu 16.0.4
**Python**: 2.7.12

%23 Exception
```
Error: Error: Traceback (most recent call last):
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 90, in main
    cli(None, None, None)
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/zbigniew/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home/command.py", line 38, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "/home/zbigniew/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "/home/zbigniew/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "/home/zbigniew/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
  File "/home/zbigniew/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "/home/zbigniew/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: /home/zbigniew/.platformio/packages/contrib-pysite/_cffi_backend.so: undefined symbol: PyUnicodeUCS4_FromUnicode

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/zbigniew/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359
	at ChildProcess.h (/home/zbigniew/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```
---------------------------------

voltane, 2019-11-02:
Solution which works for me:
It seems that PlatformIO needs Python compiled with `--enable-unicode=ucs4`.
To check if Your python is compiled with that option, simply run `python` in terminal and:
`
>>> import sys
>>> print sys.maxunicode
`
If it prints `65535` than Your Python is compiles with UCS2. If it prints out `1114111` than You already have Python which UCS4 and my solution will not work for You (probably).

To get new python You have to compile it by Yourself for example from this simple [tutorial](https://tecadmin.net/install-python-2-7-on-ubuntu-and-linuxmint/).
Keep in mind to change the configure step as follow:
`sudo ./configure --enable-optimizations --enable-unicode=ucs4`.

After that my Python upgraded from 2.7.12 to 2.7.16 and not PlatformIO works fine in VSC.

ivankravets, 2019-11-21:
Please upgrade to the latest 1.10 version of this extension. We added better checking for a valid Python.

In any case, it's better to remove broken Python from a system and use the latest official Python 3.5+

closed 2019-11-21 by ivankravets

voltane, 2019-11-22:
Well, Python without UCS4 is still a valid one.
You should add a hint about that case in docs to prevent such problems in future.