platformio/platformio-vscode-ide #1328: Start PIO Home Server -- fails on maOS

CarlSorensen, 2019-11-02:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.3
System: Darwin, 16.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Users/carlsorensen/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359
	at ChildProcess.h (/Users/carlsorensen/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

CarlSorensen, 2019-11-02:
I did lots of things to try to get it to work.  I'm not sure which things were important.  But here are some pieces:

Install python with Homebrew (in response to UCC4 problem
Alias Homebrew python3 to python
Install platformio with Homebrew
rm -rf ~/.platformio/penv  then  rm -rf ~/.vscode then restart VSCode

The last is the one that worked finally.  I think that it may have worked sooner had I done the 
`rm -rf ~/.platrformio/penv && rm -rf ~/.vscode`

I would suggest the above command as the first thing to try, and as something to try every time you have something you think might fix your problem (e.g. new Python version).

jake653, 2019-11-03:
Ditto 

Same error experienced in my new install attempt.

Note: Apr 19 "Fix" note by ivankravets:
> This is a bug with our CDN and has been just fixed. Please Restart VSCode and re-run $ pio  update. Sorry.

https://github.com/platformio/platformio-core/issues/2341
--

I can just add the dumb observation:
/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js
does not contain: contrib-pysite

ESDCore, 2019-11-05:
I had a similar issue on both VSCode and Atom:

This was eventually very easily resolved by doing the the following:

- Downloaded the Python3.8.0 installer from python.org
- Installed the root certificates as requested by the installer 
- Relaunched VSCode

Hope this helps

farriermatt, 2019-11-09:
yep, i give up i will put it on windows

ivankravets, 2019-11-16:
Could you open VSCode > Help > Toggle Developer Tools > Console and provide an output here?

CarlSorensen, 2019-11-16:
Unfortunately, I don’t think it will help to do so, because it’s now working for me.

Carl

closed 2019-11-16 by ivankravets