platformio/platformio-vscode-ide #1333: Off-screen dialog

webtop, 2019-11-03:
Issue Type: <b>Bug</b>

Create a new PIO project
Uncheck the default save location checkbox
Dialog extends off the bottom of the screen (I guess depending on content)
Dialog header does not allow it to be moved

(see https://1drv.ms/u/s!Arg9f0GH8EvIhZQAlqsoTMu93Zi0Xw?e=fclj5H)
Not a huge deal, but might annoy some newer users.

Thanks

Extension version: 1.9.3
VS Code version: Code 1.39.2 (6ab598523be7a800d7f3eb4d92d7ab9a66069390, 2019-10-15T15:35:18.241Z)
OS version: Windows_NT x64 10.0.18363

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-3770K CPU @ 3.50GHz (8 x 3606)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_deferred_display_list: disabled_off<br>skia_renderer: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|15.97GB (8.48GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

ivankravets, 2019-11-21:
Could you reproduce this issue with the latest PIO Core and extension?

webtop, 2019-11-22:
Yes, same. When unchecking the location checkbox.
I have a number of drives, so it expands to show them, and there is where the issue is.