platformio/platformio-vscode-ide #1336: Start PIO Home Server

feurig, 2019-11-06:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.3
System: Darwin, 15.6.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Users/don/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359
	at ChildProcess.h (/Users/don/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-11-16:
Could you open VSCode > Help > Toggle Developer Tools > Console and provide an output here?

feurig, 2019-11-17:
Sorry my first interaction with vscode crashed and offered to file a bug. If that automation didn't provide enough information info I am sorry. Going back to atom now.....

feurig, 2019-11-17:
[Extension Host] debugger listening on port 35092
extensionService.ts:802 Couldn't find message for key python.snippet.launch.unitTests.description.
_logMessageInConsole @ extensionService.ts:802
console.ts:134 [Extension Host] runCommand platformio Array(1) Object
console.ts:134 [Extension Host] ApplicationInsights:CorrelationIdManager Array(1)
t.log @ console.ts:134
console.ts:134 [Extension Host] ApplicationInsights:CorrelationIdManager Array(1)
t.log @ console.ts:134
console.ts:134 [Extension Host] (electron) The 'lstatSyncNoException' function has been deprecated and marked for removal.
t.log @ console.ts:134
console.ts:134 [Extension Host] runCommand platformio Array(4) Object
console.ts:134 [Extension Host] Found PIO Core 4.1.0
console.ts:134 [Extension Host] runCommand platformio Array(7) Object
2console.ts:134 [Extension Host] (electron) The 'lstatSyncNoException' function has been deprecated and marked for removal.
t.log @ console.ts:134

closed 2019-11-17 by feurig