platformio/platformio-vscode-ide #1337: Start PIO Home Server

YHare, 2019-11-06:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.3
System: Darwin, 15.6.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Users/Yuri/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359
	at ChildProcess.h (/Users/Yuri/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-11-16:
Could you open VSCode > Help > Toggle Developer Tools > Console and provide an output here?

ivankravets, 2019-11-20:
1. Please [upgrade PlatformIO IDE for VSCode to 1.10.0](https://community.platformio.org/t/platformio-ide-for-vscode-1-10/10676) or above
2. Remove `YOUR_USER_HOME_DIR/.platformio/penv` folder
3. Restart VSCode.

If you have a problem with Python interpreter, please uninstall existing [install the latest official Python 3](http://docs.platformio.org/en/latest/faq.html#install-python-interpreter).

closed 2019-11-20 by ivankravets