platformio/platformio-vscode-ide #1338: Start PIO Home Server

AkhileshThorat, 2019-11-06:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.3
System: Darwin, 16.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359
	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-11-16:
Could you open VSCode > Help > Toggle Developer Tools > Console and provide an output here?

AkhileshThorat, 2019-11-16:
[Extension Host] activating extension
console.ts:137 [Extension Host] starting language server
console.ts:137 [Extension Host] (node:67211) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(1) Object
console.ts:137 [Extension Host] runCommand platformio Array(4) Object
console.ts:137 [Extension Host]  Error: Can not retrieve the latest PlatformIO version

t.log @ console.ts:137
console.ts:137 [Extension Host] Found PIO Core 4.0.3
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:137
notificationsAlerts.ts:40 Error: Error: Detected unknown package 'contrib-pysite'  	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359 	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292) 	at ChildProcess.emit (events.js:182:13) 	at maybeClose (internal/child_process.js:962:16) 	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
onDidNotificationChange @ notificationsAlerts.ts:40
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292)	at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ console.ts:137
notificationsAlerts.ts:40 Error: Error: Detected unknown package 'contrib-pysite'  	at /Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:10359 	at ChildProcess.h (/Users/macbook/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:6292) 	at ChildProcess.emit (events.js:182:13) 	at maybeClose (internal/child_process.js:962:16) 	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

ivankravets, 2019-11-16:
Could you now open PlatformIO IDE Terminal and type `pio home`? Do you see any errors in the terminal?

AkhileshThorat, 2019-11-16:
> Could you now open PlatformIO IDE Terminal and type `pio home`? Do you see any errors in the terminal?

MacBooks-MacBook-Pro:~ macbook$ pio home
CorePackageManager: Installing contrib-pysite @ ~2.37.190418
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
contrib-pysite @ 2.37.191017 has been successfully installed!
CorePackageManager: Installing contrib-piohome @ ^2.3.2
Downloading  [####################################]  100%          
contrib-piohome @ 2.3.3 has been successfully installed!

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

Open PIO Home in your browser by this URL => http://127.0.0.1:8008
PIO Home has been started. Press Ctrl+C to shutdown.

ivankravets, 2019-11-16:
Does it work now from VSCode if you restart IDE?

AkhileshThorat, 2019-11-16:
> Does it work now from VSCode if you restart IDE?

Error still persists

ivankravets, 2019-11-16:
Please email me@ikravets.com for the further steps. Thanks!

jakematlick, 2019-11-18:
Hi @ivankravets were you able to resolve this issue? I am seeing the same thing.

ivankravets, 2019-11-19:
@jakematlick what is your OS? Version?

jakematlick, 2019-11-19:
macOS 10.14.5

ivankravets, 2019-11-19:
Could you run this command in system terminal?
```
~/.platformio/penv/bin/python --version
```

jakematlick, 2019-11-19:
Python 3.5.1

ivankravets, 2019-11-19:
Now please
```
~/.platformio/penv/bin/python -c "import urllib.request; print(urllib.request.urlopen('https://github.com').read())"
```

Do you see output?

jakematlick, 2019-11-19:
`Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/urllib/request.py", line 1240, in do_open
    h.request(req.get_method(), req.selector, req.data, headers)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/http/client.py", line 1083, in request
    self._send_request(method, url, body, headers)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/http/client.py", line 1128, in _send_request
    self.endheaders(body)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/http/client.py", line 1079, in endheaders
    self._send_output(message_body)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/http/client.py", line 911, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/http/client.py", line 854, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/http/client.py", line 1237, in connect
    server_hostname=server_hostname)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 376, in wrap_socket
    _context=self)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 747, in __init__
    self.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 983, in do_handshake
    self._sslobj.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 628, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:645)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/urllib/request.py", line 162, in urlopen
    return opener.open(url, data, timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/urllib/request.py", line 465, in open
    response = self._open(req, data)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/urllib/request.py", line 483, in _open
    '_open', req)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/urllib/request.py", line 443, in _call_chain
    result = func(*args)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/urllib/request.py", line 1283, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/urllib/request.py", line 1242, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:645)>`

jakematlick, 2019-11-20:
```
~/.platformio/penv/bin/python -c "import _ssl; print(_ssl.OPENSSL_VERSION)"
OpenSSL 0.9.8zh 14 Jan 2016
```

ivankravets, 2019-11-20:
1. Please [upgrade PlatformIO IDE for VSCode to 1.10.0](https://community.platformio.org/t/platformio-ide-for-vscode-1-10/10676) or above
2. Remove `YOUR_USER_HOME_DIR/.platformio/penv` folder
3. Restart VSCode.

If you have a problem with Python interpreter, please uninstall existing [install the latest official Python 3](http://docs.platformio.org/en/latest/faq.html#install-python-interpreter).

closed 2019-11-20 by ivankravets

jakematlick, 2019-11-20:
This resolved the issue. Thank you.

noumanqaiser, 2020-01-07:
HI @ivankravets , I am facing the same issue. I have VScode 1.41 and installed platformIO IDE 1.10 on my Windows 10 Laptop. I tried deleting the /penv folder aswell as removing PlatformIO and installing again but unfortunately I still always get the error "detected unknown package 'contrib-pysite'". In your opinion what could be causing this issue.