platformio/platformio-vscode-ide #1341: Installation Manager

Timka654, 2019-11-07:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.3
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\relif\.platformio\.cache\tmp-16060UaRXzl4vV8D9\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\relif\.platformio\.cache\tmp-16060UaRXzl4vV8D9\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\relif\.platformio\.cache\tmp-16060UaRXzl4vV8D9\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\relif\.platformio\.cache\tmp-16060UaRXzl4vV8D9\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\relif\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\relif\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\relif\\.platformio\\python37'
New python executable in C:\Users\relif\.platformio\penv\Scripts\python.exe

    at t.value (c:\Users\relif\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\installer\stages\platformio-core.js:303:14)
```

ivankravets, 2019-11-16:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1213. See solution https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-11-16 by ivankravets

Timka654, 2019-11-18:
Problem with UAC Windows 10 and python(installed in system after install PlatformIO), i'm set admin right for Visual Studio Code and use this manual http://docs.platformio.org/en/latest/installation.html#virtual-environment, this solve problem