platformio/platformio-vscode-ide #1346: Start PIO Home Server

WHO-CR, 2019-11-08:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.40.0
PIO IDE: v1.9.3
System: Windows_NT, 10.0.17134, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:10266
	at ChildProcess.h (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:6199)
	at ChildProcess.emit (events.js:200:13)
	at ChildProcess.cp.emit (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
```

ivankravets, 2019-11-16:
Could you open VSCode > Help > Toggle Developer Tools > Console and provide an output here?

WHO-CR, 2019-11-18:
Hi Ivan. I don't know if the following paste is this what you expected. If not, tell me how I can provide the right output. Sorry, but I'm a beginner in VSCode and platformio.

console.ts:137 [Extension Host] Git installation not found.
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(1) Object
console.ts:137 [Extension Host] runCommand platformio Array(4) Object
console.ts:137 [Extension Host] Found PIO Core 4.1.0
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'
	at c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:10266	at ChildProcess.h (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:6199)	at ChildProcess.emit (events.js:200:13)	at ChildProcess.cp.emit (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)	at maybeClose (internal/child_process.js:1021:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'
	at c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:10266	at ChildProcess.h (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:6199)	at ChildProcess.emit (events.js:200:13)	at ChildProcess.cp.emit (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)	at maybeClose (internal/child_process.js:1021:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
t.log @ console.ts:137
console.ts:137 [Extension Host] runCommand platformio Array(7) Object
console.ts:137 [Extension Host] Error: Error: Detected unknown package 'contrib-pysite'
	at c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:10266	at ChildProcess.h (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:6199)	at ChildProcess.emit (events.js:200:13)	at ChildProcess.cp.emit (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)	at maybeClose (internal/child_process.js:1021:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
t.log @ console.ts:137
notificationsAlerts.ts:40 Error: Error: Detected unknown package 'contrib-pysite'  	at c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:10266 	at ChildProcess.h (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:6199) 	at ChildProcess.emit (events.js:200:13) 	at ChildProcess.cp.emit (c:\Users\zowho\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29) 	at maybeClose (internal/child_process.js:1021:16) 	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
onDidNotificationChange @ notificationsAlerts.ts:40

ivankravets, 2019-11-18:
What is your Python version?
```
python --version
```

ivankravets, 2019-11-20:
1. Please [upgrade PlatformIO IDE for VSCode to 1.10.0](https://community.platformio.org/t/platformio-ide-for-vscode-1-10/10676) or above
2. Remove `YOUR_USER_HOME_DIR/.platformio/penv` folder
3. Restart VSCode.

If you have a problem with Python interpreter, please uninstall existing [install the latest official Python 3](http://docs.platformio.org/en/latest/faq.html#install-python-interpreter).

closed 2019-11-20 by ivankravets