platformio/platformio-vscode-ide #1347: Error: Detected unknown package 'contrib-pysite'

InnovativeIdeas, 2019-11-09:
I loaded latest version of VSCode , running on MAC High Sierra , installed platformio IDE 1.9.3

keep getting 

Error: Detected unknown package 'contrib-pysite'

tried all the workaround suggested here , removed pen , etc , no luck , cant find workaround , help

ivankravets, 2019-11-16:
Could you open VSCode > Help > Toggle Developer Tools > Console and provide an output here?

ivankravets, 2019-11-20:
1. Please [upgrade PlatformIO IDE for VSCode to 1.10.0](https://community.platformio.org/t/platformio-ide-for-vscode-1-10/10676) or above
2. Remove `YOUR_USER_HOME_DIR/.platformio/penv` folder
3. Restart VSCode.

If you have a problem with Python interpreter, please uninstall existing [install the latest official Python 3](http://docs.platformio.org/en/latest/faq.html#install-python-interpreter).

closed 2019-11-20 by ivankravets