platformio/platformio-vscode-ide #1353: pio home crashed: ModuleNotFoundError

Remering, 2019-11-11:
Error messages:
```
Error: Traceback (most recent call last):
  File "D:\msys64\mingw64\lib\python3.8\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "D:\msys64\mingw64\lib\python3.8\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "D:\msys64\mingw64\lib\python3.8\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "D:\msys64\mingw64\lib\python3.8\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "D:\msys64\mingw64\lib\python3.8\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "D:\msys64\mingw64\lib\python3.8\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "D:\msys64\mingw64\lib\python3.8\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "D:\msys64\mingw64\lib\python3.8\site-packages\platformio\commands\home\command.py", line 52, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\mafumafu\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "C:\Users\mafumafu\.platformio\packages\contrib-pysite\autobahn\twisted\wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "C:\Users\mafumafu\.platformio\packages\contrib-pysite\autobahn\wamp\auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
  File "C:\Users\mafumafu\.platformio\packages\contrib-pysite\cryptography\hazmat\primitives\kdf\pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "C:\Users\mafumafu\.platformio\packages\contrib-pysite\cryptography\hazmat\primitives\constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ModuleNotFoundError: No module named 'cryptography.hazmat.bindings._constant_time'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
Operation system: Windows 10
Python intepreter: 3.8.0 [GCC 9.2.0 64 bit (AMD64)] on win 32
**I promise the version of platformio is latest!**
Proved by `pip`:
```
 λ pip install platformio -U                                                                                                                                                                  
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple                                                                                                                                 
Requirement already up-to-date: platformio in d:\msys64\mingw64\lib\python3.8\site-packages (4.1.0)                                                                                          
Requirement already satisfied, skipping upgrade: colorama in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (0.4.1)                                                         
Requirement already satisfied, skipping upgrade: semantic-version<3,>=2.8.1 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (2.8.2)                                       
Requirement already satisfied, skipping upgrade: bottle<0.13 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (0.12.17)                                                    
Requirement already satisfied, skipping upgrade: requests<3,>=2.4.0 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (2.22.0)                                              
Requirement already satisfied, skipping upgrade: pyserial!=3.3,<4,>=3 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (3.4)                                               
Requirement already satisfied, skipping upgrade: marshmallow<3,>=2.20.5 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (2.20.5)                                          
Requirement already satisfied, skipping upgrade: click<8,>=5 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (7.0)                                                        
Requirement already satisfied, skipping upgrade: tabulate<1,>=0.8.3 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (0.8.5)                                               
Requirement already satisfied, skipping upgrade: pyelftools<1,>=0.25 in d:\msys64\mingw64\lib\python3.8\site-packages (from platformio) (0.25)                                               
Requirement already satisfied, skipping upgrade: chardet<3.1.0,>=3.0.2 in d:\msys64\mingw64\lib\python3.8\site-packages (from requests<3,>=2.4.0->platformio) (3.0.4)                        
Requirement already satisfied, skipping upgrade: idna<2.9,>=2.5 in d:\msys64\mingw64\lib\python3.8\site-packages (from requests<3,>=2.4.0->platformio) (2.8)                                 
Requirement already satisfied, skipping upgrade: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in d:\msys64\mingw64\lib\python3.8\site-packages (from requests<3,>=2.4.0->platformio) (1.25.6)     
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in d:\msys64\mingw64\lib\python3.8\site-packages (from requests<3,>=2.4.0->platformio) (2019.9.11)                       
```
Proved by `pio update`
```
λ pio upgrade                                              
You're up-to-date!                                         
PlatformIO 4.1.0 is currently the newest version available.
```
It seems like the absence of `cryptography`, but there's no use after I install it via pip.
```
λ pip install cryptography
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Requirement already satisfied: cryptography in d:\msys64\mingw64\lib\python3.8\site-packages (2.8)
Requirement already satisfied: six>=1.4.1 in d:\msys64\mingw64\lib\python3.8\site-packages (from cryptography) (1.13.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in d:\msys64\mingw64\lib\python3.8\site-packages (from cryptography) (1.13.2)
Requirement already satisfied: pycparser in d:\msys64\mingw64\lib\python3.8\site-packages (from cffi!=1.11.3,>=1.8->cryptography) (2.19)
```

ivankravets, 2019-11-11:
This is actually our bug, I don't know why we allowed our internal installer to use Python 3 from msys64/mingw. The libraries are incompatible. The fix will be soon.

jcw, 2019-11-13:
Yep, same here. Installed pio long ago, did pio update, it's now at 4.1.0, I now get this (macOS):

```
Updating espressif32                     @ f2f3285        Error: Traceback (most recent call last):
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
```

I'm ok with moving to py3 if that helps, but I'm not sure what steps to take from where I am now. Being on macOS, I would now (slightly) prefer going through homebrew, instead of using `pip`.

ivankravets, 2019-11-13:
@jcw what is your issue? Could you provide a full exception output?

jcw, 2019-11-13:
Sure, sorry - I'm not using VSCode, so probably my note was misplaced. Here's the output:

```
mackey:~ jcw$ pio update
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating tool-pioplus                    @ 2.6.0          [Up-to-date]
Updating contrib-pysite                  @ 2.27.191017    [Up-to-date]
Updating contrib-piohome                 @ 3.0.1          [Up-to-date]
Updating tool-scons                      @ 2.20501.190826 [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.15.0         [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.190710 [Up-to-date]
Updating framework-arduinoavr            @ 4.1.2          [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ 1.11.1         [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating tool-openocd-esp32              @ 1.1000.20190708[Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10004.191002 [Up-to-date]
Updating framework-espidf                @ 3.30300.190916 [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ f2f3285        Error: Traceback (most recent call last):
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/commands/update.py", line 53, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py", line 402, in platform_update
    pm.update(platform, only_packages=only_packages, only_check=only_check)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 133, in update
    BasePkgManager.update(self, pkg_dir, requirements, only_check)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 797, in update
    latest = self.outdated(pkg_dir, requirements)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 626, in outdated
    pkg_dir, manifest["__src_url"], silent=True
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/vcsclient.py", line 46, in newClient
    src_dir, remote_url, tag, silent
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/vcsclient.py", line 61, in __init__
    self.check_client()
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/vcsclient.py", line 124, in check_client
    return VCSClientBase.check_client(self)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/vcsclient.py", line 67, in check_client
    self.get_cmd_output(["--version"])
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/vcsclient.py", line 110, in get_cmd_output
    result = exec_command(args, **kwargs)
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/proc.py", line 131, in exec_command
    get_locale_encoding() or get_filesystem_encoding()
  File "/Users/jcw/.platformio/penv/lib/python2.7/site-packages/platformio/compat.py", line 35, in get_locale_encoding
    return locale.getdefaultlocale()[1]
  File "/Users/jcw/.platformio/penv/lib/python2.7/locale.py", line 545, in getdefaultlocale
    return _parse_localename(localename)
  File "/Users/jcw/.platformio/penv/lib/python2.7/locale.py", line 477, in _parse_localename
    raise ValueError, 'unknown locale: %s' % localename
ValueError: unknown locale: UTF-8

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
mackey:~ jcw$
```

ivankravets, 2019-11-13:
@jcw your issue is different. Python can't determine OS locale. I've just fixed it in https://github.com/platformio/platformio-core/commit/728fd7f5b9e53f3d6e57036726f848bb6967e315

Please re-run `pio upgrade --dev` in IDE.

jcw, 2019-11-13:
Ok, my apologies for the noise, and as always: thanks, hats off for your quick & effective support.

ivankravets, 2019-11-13:
Happy coding with PlatformIO! ;)

ivankravets, 2019-11-16:
@Remering did you resolve this issue?

Remering, 2019-11-17:
Not yet. I change the `MSVC` version of Python and everything goes well. 

PS:
`NumPy` also unsupports the `GCC` version of Python. So I think if it's hard to fix it just force users not to use the `GCC` version of Python.

ivankravets, 2019-11-18:
I see, this is not official Python installation. We fill fix it.

closed 2019-11-20 by ivankravets

ivankravets, 2019-11-20:
1.10 is out! https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.10.0

Please remove `HOME_DIR/.platformio/penv` folder after update to 1.10.0 and restart VSCode.

zymil, 2019-12-28:
Guys I need some help please!

Error: Traceback (most recent call last):
  File "D:\Python27\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "D:\Python27\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "D:\Python27\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "D:\Python27\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "D:\Python27\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "D:\Python27\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "D:\Python27\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "D:\Python27\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "D:\Python27\lib\site-packages\platformio\commands\init.py", line 101, in cli
    pg.generate()
  File "D:\Python27\lib\site-packages\platformio\ide\projectgenerator.py", line 134, in generate
    contents = self._render_tpl(tpl_path, tpl_vars)
  File "D:\Python27\lib\site-packages\platformio\ide\projectgenerator.py", line 139, in _render_tpl
    return bottle.template(fs.get_file_contents(tpl_path), **tpl_vars)
  File "D:\Python27\Scripts\bottle.py", line 3630, in template
    return TEMPLATES[tplid].render(kwargs)
  File "D:\Python27\Scripts\bottle.py", line 3419, in render
    self.execute(stdout, env)
  File "D:\Python27\Scripts\bottle.py", line 3406, in execute
    eval(self.co, env)
  File "<string>", line 30, in <module>
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc6 in position 11: ordinal not in range(128)

============================================================

zymil, 2019-12-28:
I have tried moving to D:\ because I thought It was related to my name like utf-8 problems "João" -> "~" but nothing works... I have also tried updating --dev, and other versions of Python27 but nothing works...
Im desperate somebody help me pls

ivankravets, 2019-12-28:
@zymil Please uninstall all Pythons from system and install the latest using this guide https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

zymil, 2020-01-03:
The problem was my name in appdata, I solved by creating another user! Thanks anyway

RR-Rico, 2020-03-19:
> 1.10 is out! https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.10.0
> 
> Please remove `HOME_DIR/.platformio/penv` folder after update to 1.10.0 and restart VSCode.

It seems to be somehow still a problem. I have installed PlatformIO from within VSCode and received the error, no module named six. Removing penv folder and restarting VSCode solved the issue for me.

> @zymil Please uninstall all Pythons from system and install the latest using this guide https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

Is installing via VSCode an option?

commit d4153c16 (2019-11-20): Skip Python from msys, mingw, emacs installations // Resolve #1353