platformio/platformio-vscode-ide #1363: Installation Manager

Nickbarbieri1, 2019-11-14:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.40.1
PIO IDE: v1.9.3
System: Linux, 5.3.0-23-generic, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/nico/.platformio/.cache/tmp-73843HdICDCgOH0K/virtualenv-16.7.5/virtualenv.py", line 39, in <module>
    import zlib
ModuleNotFoundError: No module named 'zlib'

	at t.value (/home/nico/.vscode/extensions/platformio.platformio-ide-1.9.3/node_modules/platformio-node-helpers/dist/index.js:1:32226)
```

ivankravets, 2019-11-16:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

Nickbarbieri1, 2019-11-18:
root@nico-HP-Spectre-x360:~# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin

root@nico-HP-Spectre-x360:~# echo $PYTHONPATH

root@nico-HP-Spectre-x360:~# python --version
Python 2.7.17rc1

root@nico-HP-Spectre-x360:~# pip --version
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)

root@nico-HP-Spectre-x360:~# pip list
Package                       Version
----------------------------- ---------
asn1crypto                    0.24.0
backports.functools-lru-cache 1.5
bcrypt                        3.1.6
certifi                       2019.9.11
chardet                       3.0.4
Click                         7.0
colorama                      0.3.7
configparser                  3.5.0b2
cryptography                  2.6.1
cycler                        0.10.0
dbus-python                   1.2.12
entrypoints                   0.3
enum34                        1.1.6
future                        0.16.0
idna                          2.8
ipaddress                     1.0.17
keyring                       18.0.1
keyrings.alt                  3.1.1
kiwisolver                    1.0.1
matplotlib                    2.2.3
mysql-connector-python        8.0.15
mysql-utilities               1.6.4
numpy                         1.16.2
olefile                       0.46
paramiko                      2.6.0
pexpect                       4.6.0
Pillow                        6.1.0
pip                           18.1
protobuf                      3.6.1
pycrypto                      2.6.1
pyelftools                    0.25
PyGObject                     3.34.0
PyNaCl                        1.3.0
pyodbc                        4.0.22
pyparsing                     2.2.0
pyserial                      3.4
pysqlite                      2.7.0
python-dateutil               2.7.3
pytz                          2019.2
pyxdg                         0.25
requests                      2.22.0
SecretStorage                 2.3.1
setuptools                    41.1.0
six                           1.12.0
subprocess32                  3.5.3
urllib3                       1.25.6
wheel                         0.32.3

ivankravets, 2019-11-19:
How did you install this Python? Did you compile it manually? it's broken => https://stackoverflow.com/questions/6169522/no-module-named-zlib

closed 2019-11-19 by ivankravets