platformio/platformio-vscode-ide #1371: PlatformIO projects list only lists env with local platform definition

TD-er, 2019-11-18:
I was re-arranging all my env definitions using `extends` in my platformio.ini to make it more clean.
I'm using PIO core 4.1.1b1

The envs which do not have a platform defined in their own scope will no longer show up in the PIO list in VS code.
These ones have their platform defined in a common definition which is referred to via `extends`.
When building from the terminal command line, they build just fine.

```
[esp32_common]
extends                   = common
platform                  = ${core_esp32_1_11_1.platform}

[env:visible_esp32]
extends                   = esp32_common
platform                  = ${esp32_common.platform}
build_flags               = ${esp32_common.build_flags}   -DPLUGIN_BUILD_CUSTOM
board                     = esp32dev
extra_scripts             = pre:pre_custom_esp32.py

[env:not_visible_esp32]
extends                   = esp32_common
build_flags               = ${esp32_common.build_flags}   -DPLUGIN_BUILD_CUSTOM
board                     = esp32dev
extra_scripts             = pre:pre_custom_esp32.py
```

ivankravets, 2019-11-20:
We have not been added support for `extends` into IDE yet. A temporary solution is to declare it directly:
```
[esp32_common]
extends                   = common
platform                  = ${core_esp32_1_11_1.platform}

[env:visible_esp32]
extends                   = esp32_common
platform                  = ${esp32_common.platform}
build_flags               = ${esp32_common.build_flags}   -DPLUGIN_BUILD_CUSTOM
board                     = esp32dev
extra_scripts             = pre:pre_custom_esp32.py

[env:not_visible_esp32]
extends                   = esp32_common
platform                  = ${esp32_common.platform}
build_flags               = ${esp32_common.build_flags}   -DPLUGIN_BUILD_CUSTOM
board                     = esp32dev
extra_scripts             = pre:pre_custom_esp32.py
```

TD-er, 2019-11-20:
Yep, that's how I now have the work-around.

ivankravets, 2020-06-02:
Resolved in https://github.com/platformio/platformio-node-helpers/commit/10f0b9406909d009a1ce337175593cf26a830e1c

closed 2020-06-02 by ivankravets