platformio/platformio-vscode-ide #1372: Start PIO Home Server

erwango, 2019-11-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.40.1
PIO IDE: v1.9.3
System: Windows_NT, 10.0.16299, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at c:\Users\frq07517\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:10266
	at ChildProcess.h (c:\Users\frq07517\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\dist\index.js:1:6199)
	at ChildProcess.emit (events.js:200:13)
	at ChildProcess.cp.emit (c:\Users\frq07517\.vscode\extensions\platformio.platformio-ide-1.9.3\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
```

erwango, 2019-11-19:
Cf what I saw in https://github.com/platformio/platformio-vscode-ide/issues/1097:

```
PS C:\Users\frq07517> .platformio/penv/Scripts/python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
Traceback (most recent call last):
  File "C:\Users\frq07517\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 662, in urlopen
    self._prepare_proxy(conn)
  File "C:\Users\frq07517\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 948, in _prepare_proxy
    conn.connect()
  File "C:\Users\frq07517\.platformio\penv\lib\site-packages\urllib3\connection.py", line 394, in connect
    ssl_context=context,
  File "C:\Users\frq07517\.platformio\penv\lib\site-packages\urllib3\util\ssl_.py", line 370, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "C:\Python37\lib\ssl.py", line 412, in wrap_socket
    session=session
  File "C:\Python37\lib\ssl.py", line 850, in _create
    self.do_handshake()
  File "C:\Python37\lib\ssl.py", line 1108, in do_handshake
    self._sslobj.do_handshake()
ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host
```

ivankravets, 2019-11-20:
1. Please [upgrade PlatformIO IDE for VSCode to 1.10.0](https://community.platformio.org/t/platformio-ide-for-vscode-1-10/10676) or above
2. Remove `YOUR_USER_HOME_DIR/.platformio/penv` folder
3. Restart VSCode.

If you have a problem with Python interpreter, please uninstall existing [install the latest official Python 3](http://docs.platformio.org/en/latest/faq.html#install-python-interpreter).

closed 2019-11-20 by ivankravets