platformio/platformio-vscode-ide #1377: cant build simplest arduino-program

LeonFretter, 2019-11-20:
I try to build the following program, using the build-command
provided by the platformio-vscode-ide
```
#include <Arduino.h>

void setup() {
  pinMode(13, OUTPUT);
}

void loop() {
  digitalWrite(13, HIGH);
}
```
My platformio.ini is pretty much untouched, the following:
```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

Unfortunately the build fails and spits out the following plethora of error-messages..
A recurring thing seems to be the un-declared `uint_farptr_t`, but there are other errors mixed in as well.
See for yourself:
```
/usr/include/math.h:283:48: note: in definition of macro '__MATHDECL_1'
   extern type __MATH_PRECNAME(function,suffix) args __THROW
                                                ^
/usr/include/bits/mathcalls-helper-functions.h:36:31: note: in expansion of macro '_Mdouble_'
 __MATHDECL_1 (int, __isnan,, (_Mdouble_ __value)) __attribute__ ((__const__));
                               ^
In file included from /usr/include/features.h:424:0,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from /home/pete/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:23,
                 from /home/pete/.platformio/packages/framework-arduinoavr/cores/arduino/Stream.cpp:25:
/usr/include/sys/cdefs.h:61:20: error: expected ',' or ';' before 'throw'
 #   define __THROW throw ()
                    ^
/usr/include/math.h:283:53: note: in expansion of macro '__THROW'
   extern type __MATH_PRECNAME(function,suffix) args __THROW
                                                     ^
/usr/include/bits/mathcalls-helper-functions.h:36:1: note: in expansion of macro '__MATHDECL_1'
 __MATHDECL_1 (int, __isnan,, (_Mdouble_ __value)) __attribute__ ((__const__));
 ^
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1560:36: error: 'uint_farptr_t' has not been declared
 extern void *memcpy_PF(void *dest, uint_farptr_t src, size_t len);
                                    ^
In file included from /home/pete/.platformio/packages/toolchain-atmelavr/avr/include/util/delay.h:46:0,
                 from /home/pete/.platformio/packages/framework-arduinoavr/cores/arduino/USBAPI.h:27,
                 from /home/pete/.platformio/packages/framework-arduinoavr/cores/arduino/PluggableUSB.cpp:20:
/usr/include/bits/mathcalls-helper-functions.h:21:36: error: '_Float128' was not declared in this scope
 __MATHDECL_1 (int, __fpclassify,, (_Mdouble_ __value))
                                    ^
In file included from /home/pete/.platformio/packages/framework-arduinoavr/cores/arduino/Arduino.h:26:0,
                 from /home/pete/.platformio/packages/framework-arduinoavr/cores/arduino/IPAddress.cpp:20:
/usr/include/math.h:415:21: warning: '__issignalingf128' initialized and declared 'extern'
 # define _Mdouble_  _Float128
                     ^
/usr/include/math.h:283:48: note: in definition of macro '__MATHDECL_1'
   extern type __MATH_PRECNAME(function,suffix) args __THROW
                                                ^
/usr/include/bits/mathcalls-helper-functions.h:42:37: note: in expansion of macro '_Mdouble_'
 __MATHDECL_1 (int, __issignaling,, (_Mdouble_ __value))
                                     ^
/usr/include/math.h:415:21: error: '_Float128' was not declared in this scope
 # define _Mdouble_  _Float128
                     ^
/usr/include/math.h:283:48: note: in definition of macro '__MATHDECL_1'
   extern type __MATH_PRECNAME(function,suffix) args __THROW
[... 844 lines cut ...]
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1665:38: error: 'uint_farptr_t' has not been declared
 extern int strcmp_PF(const char *s1, uint_farptr_t s2) __ATTR_PURE__;
                                      ^
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1682:39: error: 'uint_farptr_t' has not been declared
 extern int strncmp_PF(const char *s1, uint_farptr_t s2, size_t n) __ATTR_PURE__;
                                       ^
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1698:42: error: 'uint_farptr_t' has not been declared
 extern int strcasecmp_PF(const char *s1, uint_farptr_t s2) __ATTR_PURE__;
                                          ^
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1716:43: error: 'uint_farptr_t' has not been declared
 extern int strncasecmp_PF(const char *s1, uint_farptr_t s2, size_t n) __ATTR_PURE__;
                                           ^
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1732:40: error: 'uint_farptr_t' has not been declared
 extern char *strstr_PF(const char *s1, uint_farptr_t s2);
                                        ^
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1744:37: error: 'uint_farptr_t' has not been declared
 extern size_t strlcpy_PF(char *dst, uint_farptr_t src, size_t siz);
                                     ^
/home/pete/.platformio/packages/toolchain-atmelavr/avr/include/avr/pgmspace.h:1760:36: error: 'uint_farptr_t' has not been declared
 extern int memcmp_PF(const void *, uint_farptr_t, size_t) __ATTR_PURE__;
                                    ^
*** [.pio/build/uno/FrameworkArduino/HardwareSerial3.cpp.o] Error 1
*** [.pio/build/uno/FrameworkArduino/HardwareSerial2.cpp.o] Error 1
*** [.pio/build/uno/FrameworkArduino/IPAddress.cpp.o] Error 1
```

ivankravets, 2019-11-20:
This is not an issue with IDE. Please ask here https://community.platformio.org/

closed 2019-11-20 by ivankravets

Vogelm90, 2022-03-19:
I have been having the same problem, and so far, this is the only comment i have found that has the exact same problem as I have. Was wondering if you have found any solutions for this?