platformio/platformio-vscode-ide #1384: Simply does not work :(

stedesDev, 2019-11-21:
HI...
How is life...

I de-installed every thing, delete everythings... 
Re-intalled Python 3.8, VS Code

And then re installed Platform IO... 

[image]

but... doesn't work...

And tried it on another computer... same result

Regards, 

Steph

ivankravets, 2019-11-22:
What is your OS? Do you use the latest VScode and PlatformIO IDE extension?

stedesDev, 2019-11-22:
Hi…

Win 10 and yep both to the latest version

Regards,

ivankravets, 2019-11-22:
Please contact me@ikravets.com for further details.

stedesDev, 2019-11-22:
> Please contact me@ikravets.com<mailto:me@ikravets.com> for further details.

HI…
thank alot to you…

I really don’t know why… every thing restart to work well, like magic…

Again thnks…

ivankravets, 2019-11-22:
Hm... Strange. In any case, it's good that it works!

Happy coding with PlatformIO!

closed 2019-11-22 by ivankravets