platformio/platformio-vscode-ide #1385: Float support for nanoatmega328 in sprintf()

giorgakos, 2019-11-21:
I am not sure if this one is a bug or not but this

```
double Voltage1 = 0.0;
sprintf(ln2, "%f Volt", Voltage1);
```

generates `? Volt` as output with the following configuration

```
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
```

ivankravets, 2019-11-22:
Please forward to https://platformio.org/

closed 2019-11-22 by ivankravets

e-tinkers, 2020-01-27:
> I am not sure if this one is a bug or not but this
> 
> ```
> double Voltage1 = 0.0;
> sprintf(ln2, "%f Volt", Voltage1);
> ```
> 
> generates `? Volt` as output with the following configuration
> 
> ```
> [env:nanoatmega328]
> platform = atmelavr
> board = nanoatmega328
> framework = arduino
> ```

Just want to point out that this has nothing to do with PlatformIO, if you run the code in Arduino IDE, it will have the same problem. This is because the Arduino implementation of `sprintf` is based on `avr libs`, which for saving the memory, does not implement the support for floating point. This is a known issue since day 1 of Arduino release. The problem can be fixed easily by allowing user to add compiler options, but Arduino's compiler options are hardcoded in its Java code, and for years it never bother to fix it or give user a choice to add custom compiler options. This is one among many reasons I use PlatformIO nowadays. 

The solution can be found in avr libc's [`vprintf`](http://www.nongnu.org/avr-libc/user-manual/group__avr__stdio.html#gaa3b98c0d17b35642c0f3e4649092b9f1) documentation:

>Since the full implementation of all the mentioned features becomes fairly large, three different 
>flavours of vfprintf() can be selected using linker options. The default vfprintf() implements all the 
>mentioned functionality except floating point conversions. 

>If the full functionality including the floating point conversions is required, the following options
>should be used:

```
-Wl,-u,vfprintf -lprintf_flt -lm
```

This can be implemented in PlatformIO by adding the `build_flags` as:

```
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
build_flags = -Wl,-u,vfprintf -lprintf_flt -lm
```

clabnet, 2021-02-16:
Thank's e-thinkers, your reply is it correct and very detailed.

pkchowdhury, 2022-04-07:
cannot find -lprintf_flt error occured. Could you provide me any soluiton.? My platformio,ini conf is given bellow.

[env:blackpill_f401cc]
platform = ststm32
board = blackpill_f401cc
framework = arduino
build_flags = -Wl,-u,vfprintf -lprintf_flt -lm
board_build.mcu = stm32f401ccu6
board_build.f_cpu = 16000000L
upload_protocol = stlink

e-tinkers, 2022-04-07:
The build-flag mentioned above is applicable only to avr-based Arduino (i.e. Uno, Nano, Pro Mini, etc.) that utilising AVRlib from Microchip. You are running stm32 which has a totally different Arduino Core. 

For stm32 (as well as ESP8266/ESP32), `Serial.printf()` is support out of box, and floating point can be supported via pulldown menu selection on Arduino IDE (tools -> C Running Liibrary -> Newlib Nano + Float Printf) or add `build_flags =  -D PIO_FRAMEWORK_ARDUINO_NANOLIB_FLOAT_PRINTF` if you are using platformio.