platformio/platformio-vscode-ide #1388: IntellieSense Broken in 4.1.1b2

fabltd, 2019-11-22:
IntellieSense show the following errors:

#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/Users/user/Documents/GitHub/test/src/main.cpp).

cannot open source file "Availability.h" (dependency of "Arduino.h")

Pio rebuild IntellieSense does not fix this.

Seems it cannot find Availability.h that is a dependency of Arduino.h 

Help.

fabltd, 2019-11-22:
removing .vscode does not fix this.  remove .vscode restart vscode problem persists.

mereshow, 2019-11-24:
Same here (PlatformIO 1.4.0 on Ubuntu 19.10). Even the blink project example fails.

fabltd, 2019-11-25:
Remove and Re install the intellieSense plug in seems to have fixed it for me

volcini, 2019-11-25:
Perhaps check #1392.

ivankravets, 2020-06-03:
It was a bug of VSCode. Please upgrade VSCode to the latest version and restart it. Sorry.

closed 2020-06-03 by ivankravets