platformio/platformio-vscode-ide #1392: Erreurs #include détectées. Mettez à jour includePath. Les tildes sont désactivés pour cette unité de traduction

themrstone24, 2019-11-23:
the issue come from the line : "#include <Arduino.h>"

volcini, 2019-11-24:
I have possibly same issue here since a few days ago.
Platform: Espressif 32
PIO Home 3.0.1, Core 4.1.0
Project can be empty one, just created from scratch.

The error is `#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit.` following by file path and name. More open files cause more errors of the same type to be displayed and counted.
On some files each include error is followed by something like:  `cannot open source file "stdbool.h" (dependency of ...) C/C++ [1696] [1,1]`. 

I tracked the problem to include path in `c_cpp_properties.json`. 
Adding three lines to win32 includePath (and saving the file) promptly eliminates the problem.
Now I don't claim they have to be exactly these lines, since I used trial and error method to tell intellisense how to find stdbool.h, but it somehow works.

The lines are:
```
"C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/",
"C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/",
"C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/lib/gcc/xtensa-esp32-elf/5.2.0/include/",
```
Sadly, this works only until VSCODE is restarted. Since it's all about an autogenerated file, I tried adding these settings to various config files (global, workspace, project), but with no success. I tried this on three computers with same results.

volcini, 2019-11-24:
Looks like the problem can be solved by manually adding these lines to `settings.json`:
```
"C_Cpp.default.systemIncludePath": [
 "C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/",
 "C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/",
 "C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/lib/gcc/xtensa-esp32-elf/5.2.0/include/",
],
```
Restart VSCode after saving the file.
I still can't find what has changed to cause this problem.

themrstone24, 2019-11-24:
Thx for your comment, I will try this. But even with these errors my program work perfectly with no issues. You added your 3 lines in "c_cpp_properties.json" or in "settings.json" ?

volcini, 2019-11-24:
Yes, I have the same problem. The "include" errors are reported, but program compiles with no problem and everything works.

I tried first with the `c_cpp_properties.json` (which gets rewritten at each restart) and later, after more trial and error, found, that adding them to `systemIncludePath` has the fortunately the same effect.
Change the `usernamedir` part in the path for the actual user folder of course.

themrstone24, 2019-11-24:
Where can I find "systemIncludePath" ?

volcini, 2019-11-24:
It is not included in the file by default, you have to add it there or follow menu like: `File/Preferences/Settings` then on the left pick "Extensions" and "C/C++" and on top choose  Workspace or *Project name* . In the middle, find somewhere the option `C_Cpp › Default: System Include Path` that says `Edit in settings.json`. Click it, it will open the relevant file.

themrstone24, 2019-11-24:
I am sorry but I don't find were I have to add these lines in the file

volcini, 2019-11-25:
Nevermind, just put these in project's  `settings.json` (mind curly brackets..) and change usernamedir to your actual user directory:
```
{
 "C_Cpp.default.systemIncludePath": [
  "C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/",
  "C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf/",
  "C:/Users/usernamedir/.platformio/packages/toolchain-xtensa32/lib/gcc/xtensa-esp32-elf/5.2.0/include/",
],
}
```
If it's gonna work, you'll notice that immediately.

dominicgermana, 2019-11-25:
This issue seems to be related to the "0.26.2-insiders2" version of the C/C++ Intellisense extension.

If I install version "0.26.1" the problem goes away, and when I let it update to "0.26.2-insiders2" it comes back.  I had to go into the Configure Extension Settings dialog and set C_Cpp: Update Channel to Default so it would stop automatically updating to the Insiders version.

volcini, 2019-11-25:
Ok, your temporary solution works. I had to uninstall the PlatformIO though, then uninstall the C/C++ Intellisense extension, then reinstall it, decline the upgrade to "0.26.2-insiders2" and reinstall PlatformIO.

It looks like the extension doesn't downgrade if PlatformIO is installed..

dominicgermana, 2019-11-25:
@volcini in the extensions view, I was able to click on the C/C++ extension gear icon and select "Install Another Version..." and choose 0.26.1.

That worked for me without having to uninstall/reinstall PlatformIO IDE (or Workbench, another extension I have that is dependent on the C/C++ extension.  YMMV, Good luck!

ivankravets, 2020-06-03:
It should be fixed in the latest VSCode version. Please check that you use the latest VSCode and extensions.

closed 2020-06-03 by ivankravets

ipa64, 2020-10-29:
I have the same issue with VsCode on Mac, I will try to edit settings.json

ipa64, 2020-10-29:
Yes, it's working, I added this path
/Users/username/.platformio/packages/toolchain-sdcc/share/sdcc/include/

ivankravets, 2020-10-29:
@ipa64 please report issue to https://github.com/platformio/platform-intel_mcs51/issues

ipa64, 2020-10-29:
Hi, ok but why intel_mcs51 ? 
I use a STM8
Thank you

ivankravets, 2020-10-29:
Sorry, please report here https://github.com/platformio/platform-ststm8/issues and will fix.