platformio/platformio-vscode-ide #1395: Start PIO Home Server

docsparks, 2019-11-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.40.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.17134, x64

  %23 Exception
  ```
  Error: Error processing line 9 of C:\Users\WorkNew\.platformio\packages\contrib-pysite\pywin32.pth:

  Traceback (most recent call last):
    File "C:\Python38\lib\site.py", line 168, in addpackage
      exec(line)
    File "<string>", line 1, in <module>
  AttributeError: module 'os' has no attribute 'add_dll_directory'

Remainder of file ignored
Error: Traceback (most recent call last):
  File "c:\users\worknew\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "c:\users\worknew\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\worknew\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\worknew\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\worknew\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\worknew\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\worknew\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\worknew\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 52, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "C:\Users\WorkNew\.platformio\packages\contrib-pysite\autobahn\twisted\__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "C:\Users\WorkNew\.platformio\packages\contrib-pysite\autobahn\twisted\wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "C:\Users\WorkNew\.platformio\packages\contrib-pysite\autobahn\wamp\auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
  File "C:\Users\WorkNew\.platformio\packages\contrib-pysite\cryptography\hazmat\primitives\kdf\pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "C:\Users\WorkNew\.platformio\packages\contrib-pysite\cryptography\hazmat\primitives\constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: DLL load failed: The specified procedure could not be found.

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\WorkNew\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\WorkNew\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:200:13)
	at ChildProcess.cp.emit (c:\Users\WorkNew\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

AUmrysh, 2019-11-27:
Same exact issue here

anugrahmasih261, 2020-05-20:
useless help

closed 2020-06-03 by ivankravets