platformio/platformio-vscode-ide #1398: 'platformio-ide.build' not found

edvinmellergard, 2019-11-24:
Hi, I'm having problem with platformio since a few days back. Haven't used VS Code for a few weeks and after updating everything I can no longer build.

I get the error: command 'platformio-ide.build' not found

I found a few people with similar issues but their solution didn't work for me.  For example: https://community.platformio.org/t/problem-with-platformio-installation-on-vscode/9983
https://github.com/platformio/platformio-vscode-ide/issues/1127

I have tried:
1. Uninstalling VS Code
2. Uninstall pyton
3. Delete %user%/.platformio 
4. Delete %user%/.vscode
5. Downlaod VScode and install
6. Install platformio extension
7. Downlaod and install latest pyton (3.8.0) and add it to PATH.

Any one have any ideas?

[image]

jhoughjr, 2019-12-08:
I'm having the same issues on macOS Catalina. Home was crashing until a reboot for a few days, probably since the last update. Now I can't do anything and my project is dead in the water.

jhoughjr, 2019-12-08:
I'm trying steps 3,4,6 will report if it works for me.

jhoughjr, 2019-12-08:
:/ didn't/t work. Guess ill have to try Atom.

adx-git, 2019-12-22:
Did you guys find a solution?
I am having the same problem.

jhoughjr, 2019-12-22:
closest I have come is to unistall the extension and just use pio command line.
Ive also started just using esp-idf

edvinmellergard, 2019-12-22:
No solution yet :( I installed VS Code on another computer and run the build on that one after syncing the whole compiled project from the first computer to the second.

Crunch91, 2020-01-24:
Same problem here. Tried same process:

1. Uninstalled PIO extension in workspace
2. Removed %USERNAME%\.vscode\extensions\platformio.platformio-ide-1.10.0 folder
3. Removed %USERNAME%\.platformio folder
4. Restarted VS Code

It keeps saying:
[image]

And the toolbar icon is also missing:
[image]

FoxExe, 2020-01-27:
Solved only when i completely uninstall Visual Code and reinstall all again.
But idk what exactly helps...

backpacked, 2020-01-30:
I had PlatformIO working when I installed it first. Then I tried to add some external libraries (Arduino) etc. to a project using sym link, which did not work and then copy, which works but who wants to copy the same libraries in multiple projects. So I removed the libraries. Then I noticed, the tools vanished and the commands don't work. The only thing that works is pio commands. The UI seems to be flaky. I tried every possible solution barring reinstalling VSCode, which I have no plans on doing as I have other extensions that are needed and working.

Crunch91, 2020-02-02:
> 
> 
> Solved only when i completely uninstall Visual Code and reinstall all again.
> But idk what exactly helps...

Did not help in my case. Still same behaviour as before.

heumann-a, 2020-03-19:
tried same solutions but got same behaviour. could not start ide at all....

FoxExe, 2020-03-25:
Uninstall VisualCode and delete all whats left:
C:\Program Files\Microsoft VS Code
C:\Users\%UserName%\.vscode
C:\Users\%UserName%\.platformio

Then, reinstall VSCode and plugins. Its help me.

owennewo, 2020-05-02:
I got the same error when I opened the folder which didn't have a platformio.ini file.  When I opened the correct (child) folder it worked.  It would be nice if platformio shouted if you were trying to build in a folder that doesn't have a platformio.ini instead of giving a cryptic 'platformio-ide.build' not found' error.

I'm not saying that OP has same problem as me but it's a possibility.

ivankravets, 2020-06-03:
Hi,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

nnukem, 2020-06-09:
It does not solve the problem. Nothing has changed.
[image]

ivankravets, 2020-06-09:
You need to open PlatformIO project. Please click on Home icon in the bottom of toolbar and press open project

Nuntis-Spayz, 2021-02-28:
I am gettin gthis "command 'platformio-ide.build' not found" on both my Windows & MacBook Pro.

Not tried to fix Mac yet, but on Windows, uninstalling and deleting all python, platformio and vscode apps and settings/folder, and also uninstalled Mu (which was used for the new Rapsberry Pi Pico), and removing all references in paths - reboot and reinstall

Still the same "command 'platformio-ide.build' not found"

Noman5237, 2021-03-21:
If nothing works rollback to version 2.1.0, and it worked for me

ivankravets, 2021-03-21:
@Noman5237 do you mean that the latest 2.3.0 does not work for you?

Noman5237, 2021-03-21:
Yeah I currently cannot make the latest version.

ivankravets, 2021-03-21:
Please switch back to the latest version and provide more details on how to reproduce your issue. 
`VSCode > Menu > Help > Toggle Developer Tools > Console` should help you.

Noman5237, 2021-03-21:
> Please switch back to the latest version and provide more details on how to reproduce your issue.
> `VSCode > Menu > Help > Toggle Developer Tools > Console` should help you.

I updated to the latest version 2.3.0 and now it works flawlessly,😕
Previously none of the build or upload buttons showed up, i couldn't even made it work from command palette.
Well no worries for now, thank you @ivankravets 🙇‍♂️

ivankravets, 2021-03-21:
Great that it works now! 

Happy coding with PlatformIO! 🚀

dontsovcmc, 2021-03-28:
I see "'platformio-ide.build' not found"
"'platformio-ide.upload' not found"
"'platformio-ide.clean' not found" errors when something is installing inside VS Code Platformio extension and I press "compile", "upload", "clean". First run.  I think it is first install framework.
Wait a minute and all works fine.

Windows 10x64

mbratch, 2021-04-09:
I have the same problem: "command platformio-ide.build" not found when attempting to build with espidf platform. I can build fine when selecting arduino platform. I'm using PlatformIO home version 3.3.4, core 5.1.1.

This is frustrating. I absolutely do not wish to uninstall all this stuff and re-install it. There must be some configuration or something somewhere which remedies this. Does anyone understand the root cause of the error?

ivankravets, 2021-04-09:
@mbratch do you see ⚙️ the indicator at the bottom "blue" panel that indicates rebuilding project index?

mbratch, 2021-04-09:
@ivankravets thanks so much for the quick response. I had just this morning suddenly run into this issue enabling the esp32 platform. After fussing with it awhile I finally decided to completely uninstall, remove, and reinstall platformio and it did clear up the problems. So looks good now. Sorry for the earlier rant.

ivankravets, 2021-04-09:
Hm, strange issue. It could be great to reproduce it.

molod4uk, 2021-05-21:
Hi, I have the same problem. I installed the extension not having Arduino IDE installed and after installing Arduino IDE problem is still here. I tried uninstalling the extension and Arduino IDE but it didn't help.

gerard-hm, 2021-07-18:
This could be a red herring, but seems to help me in those situations is just giving the IDE a few minutes (to index the project?) and then things just magically start working. 

Sorry if this doesn't help anyone else, just wanted to share my experience with this issue...

mbratch, 2021-07-18:
Yes, I’ve noticed it needs a bit of time to get ready and there may not be any visual indication in the last several seconds that it is, indeed, not yet ready. Sent from Mail for Windows 10 From: gerard-hmSent: Sunday, July 18, 2021 5:04 PMTo: platformio/platformio-vscode-ideCc: Mark Bratcher; MentionSubject: Re: [platformio/platformio-vscode-ide] 'platformio-ide.build' not found (#1398) This could be a red herring, but seems to help me in those situations is just giving the IDE a few minutes (to index the project?) and then things just magically start working.Sorry if this doesn't help anyone else, just wanted to share my experience with this issue...

cinderblock, 2021-09-03:
100% of the time I restart VS Code, PlatformIO v2.3.3 does not seem to "activate" itself. The best method I've found to reliably get it working again is to **disable and then re-enable the extension**. They PlatformIO rebuilds the index, and then commands work as expected.

killerkalamari, 2021-09-18:
> 100% of the time I restart VS Code, PlatformIO v2.3.3 does not seem to "activate" itself. The best method I've found to reliably get it working again is to **disable and then re-enable the extension**. Then PlatformIO rebuilds the index, and then commands work as expected.

Thank you, this worked for me. I disabled the PIO extension, reloaded vscode (per prompt), then enabled PIO, and I could build.

cinderblock, 2021-09-18:
I would be more than happy to spend some time debugging this on my systems that seem to reliably exhibit this issue. I just don't know where else to look.

I've checked
- all of VS Code's OUTPUT tabs
- Developer console

@killerkalamari I have not found the "Reload VS Code" step to be needed in my experience.

Selene53, 2021-11-14:
Unfortunately using the "Disable Live Share" did not work. Seemed like a logical attempt. 

**_HOWEVER_** when I removed the the folder from the workspace and Opened the project from the PlatformIO section (as I had done before) this time it worked but the Live Share extensions were off this time. I will turn them back on and see if it fails

Thanks

cinderblock, 2021-11-16:
> Unfortunately using the "Disable Live Share" did not work. Seemed like a logical attempt.

Note, "Live Share" was only the problem *for me* because it took a lot time to start on my system. Do you have other packages with a long startup time? The "Extensions" tab  of VS Code shows you how much time each extension added to the startup (only populated/displayed after VS Code has fully finished starting up).

killerkalamari, 2022-05-11:
I encountered this problem due to having multiple folders open in my workspace. I closed all folders except for the PlatformIO project, restarted vscode (perhaps unnecessarily), and then I was able to build.

wilman-labs, 2022-07-24:
I also just got this. This time round was after a reboot (for an unrelated reason) running VS Code on Mac. 

When VS Code opened up again PlatformIO wouldn't build or upload. Tried dis/enabling, rolling back to a previous PlatformIO ver both of which didnt help. 

Finally figured that after the reboot VS Code had opened up the main file I had been working on in /src as an individual file. After closing that and reopening the PlatformIO Project that the file was part of everything worked perfectly again.

lisbravo, 2022-08-08:
Same here, sporadic behavior

benreu, 2022-09-10:
I'm moving from Atom to Visual Studio and ran into this problem immediately using the latest software. Very disappointing when PlatformIO and Microsoft force you to use a buggy program. Wait a minute, did I say Microsoft? I believe that may be some of the answer...

sahlex, 2022-09-23:
In the project directory, I submitted the following comand: 
```bash
platformio run -t clean -v
```
This gave me:
```
Processing sparkfun_redboard_turbo (platform: atmelsam; board: sparkfun_redboard_turbo; framework: arduino; lib_deps: azure/Azure SDK for C@^1.0.0, knolleary/PubSubClient@^2.8, arduino-libraries/ArduinoECCX08@^1.3.6, agdl/Base64@^1.0.0, plageoj/UrlEncode@^1.0.0, hideakitai/MQTTPubSubClient@^0.1.3, links2004/WebSockets@^2.3.7, gilmaimon/ArduinoWebsockets@^0.5.3, khoih-prog/EthernetWebServer@^2.2.3)
--------------------------------------------------------------------------------------------------------------------------------------------
Build environment is clean
Done cleaning
======================================================= [SUCCESS] Took 2.58 seconds =======================================================
```
After this, everything worked like charm.

benreu, 2022-09-23:
Update: After fiddling a while, I discovered the blue PlatformIO task bar at the bottom of VS Code saying "loading tasks". I finally walked away for several hours. When I came back, the loading tasks had gone away and everything worked.

dimasw12, 2022-09-25:
[image]
I already have the same problem, but after I try to check "use built-in platformIO core" =>restart and wait for the download to be done, all is back to normal. all commands appear now,
maybe this trick can help you.
thank you.
happy coding with platformIO

ivankravets, 2022-12-11:
Fixed in the latest development version (upcoming PlatformIO IDE 3.0). Please re-test with the pre-release version. See instruction below:

[image]

logana60, 2022-12-11:
Thank you Mr. Kravets for taking a look at this issue. Version 2.99.0 definitely has improved the issue; however, the issue is still present for me.

[image]

Now the platformio icon shows up for me in the toolbar but after a few hours the extension still says initializing like the above picture.

While clicking on different pages of vscode will get the following:
`Error loading webview: Error: Could not register service workers: InvalidStateError: Failed to register a ServiceWorker: The document is in an invalid state..`

This install of vscode only has c++ version 1.13.7 and v2.99.0 of platformio

testing different permutations of the following did not seem to help either:
- use built in core
- dev core
- built in python

Dodo17sky, 2023-01-03:
I had the same issue.
For me, I think the **Remote - SSH** extension created the problem. I just disabled the **Remote - SSH**, reload the VSCode and then it worked

cinderblock, 2023-01-03:
I'm pretty sure this issue is caused by other VS Code extensions taking a long time to startup. I've tried creating a simple extension that just does this startup delay intentionally to help debug it but that proved more difficult to make the extension load than I expected.

ivankravets, 2023-01-04:
Could you reproduce this issue with [PlatformIO IDE 3.0 Preview](https://community.platformio.org/t/platformio-ide-3-0-preview-for-vscode-we-want-your-feedback/31151/)?

Muzosh, 2023-03-24:
I'm having this issue with PlatformIO version 3.1.1. I will try to wait a little bit as well. But the issue was NOT resolved in 3.0 Preview

ivankravets, 2023-03-24:
@Muzosh , could you file a new issue? We should improve the UX and inform developers that config is broken even when developers did mistake. Thanks!

rahji, 2023-05-23:
> I'm having this issue with PlatformIO version 3.1.1. I will try to wait a little bit as well. But the issue was NOT resolved in 3.0 Preview
> 
> EDIT: I had syntax error in platformio.ini file. After correction, it started downloading and configuring bunch of stuff and it started working. You can debug the error in VSCode OUTPUT window under "PlatformIO: Project Config".

This was my issue as well - I had a duplicated line in the file. @Muzosh please do file an issue. This is something that can be really frustrating for a user and is easily fixed by improving the UX.

Muzosh, 2023-05-23:
@ivankravets I'm sorry for late response.
@rahji I think you should be the one creating this issue if you feel like it. I personally don't view it as issue when the error can be seen in the Output window. Plus, I don't even remember what this problem was about so I wouldn't be able to write a great description. You experienced this recently so you have all the details.

ivankravets, 2023-05-23:
@Muzosh, this is another bug. Could I ask you to open a new issue at https://github.com/platformio/platformio-vscode-ide/issues ?

Muzosh, 2023-05-23:
Done https://github.com/platformio/platformio-vscode-ide/issues/3723

DoniyorHalilov, 2024-06-28:
I had same error (ubuntu). I just delete .pio and uninstall platformio extension and reinstall it then everything works well

digital-slave, 2025-12-27:
if i plugin the device the problems goes away

commit e5f05fd0 (2022-12-11): Fixed an issue with "'platformio-ide.build' not found" // Resolve #1398