platformio/platformio-vscode-ide #1400: Installation Manager

cosminiq, 2019-11-25:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.40.1
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\40766\.platformio\.cache\tmp-16280AguTP6624x47\virtualenv-16.7.7\virtualenv.py", line 2628, in <module>
    main()
  File "C:\Users\40766\.platformio\.cache\tmp-16280AguTP6624x47\virtualenv-16.7.7\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\40766\.platformio\.cache\tmp-16280AguTP6624x47\virtualenv-16.7.7\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\40766\.platformio\.cache\tmp-16280AguTP6624x47\virtualenv-16.7.7\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\40766\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\40766\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\40766\\.platformio\\python37'
New python executable in C:\Users\40766\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\40766\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:34077)
  ```

ivankravets, 2019-11-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-11-28 by ivankravets

cosminiq, 2019-11-28:
eroor rezolved! my antivirus block platformIO.

ivankravets, 2019-11-28:
What is your antivirus tool?

cosminiq, 2019-11-28:
bitdefender / protection for modifing files!