platformio/platformio-vscode-ide #1406: Installation Manager

Angel215i, 2019-11-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues

  %23 Configuration

  VSCode: 1.40.2
  PIO IDE: v1.10.0
  System: Windows_NT, 10.0.18362, x64

  %23 Exception
  ```
  Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00004154 (most recent call first):

	at t.value (c:\Users\Angel\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:34077)
  ```

ivankravets, 2019-11-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1404

closed 2019-11-28 by ivankravets

Angel215i, 2019-12-17:
I keep getting errors with Marlin and installing
buildroot/vscode/autobuildmarlin
Im following directions on site to install marlin i Keep getting error
after error.  Ask i want is to update my 3d printer and print please help

ivankravets, 2019-12-21:
Try to remove all Python installations from a system and install the latest following this guide https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

Angel215i, 2019-12-22:
Ok thank you

Angel215i, 2020-01-25:
Hello
Please help i cannot compile anything ive been trying and all i think I'm
doing is messing it up.  I tried taking in what you were doing so i can
learn but i cannot compile.  I cannot use the delete trash tab i cannot use
the check mark. Ive been watching videos on YouTube and still there's
nothing that shows what I'm going through to help. Please help me out