platformio/platformio-vscode-ide #1410: AttributeError: module 'enum' has no attribute 'IntFlag'

aho-ops, 2019-11-30:
Issue Type: <b>Bug</b>

  File "C:\Users\User\.platformio\python37\lib\re.py", line 143, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
*** [upload] Error 1

Extension version: 1.10.0
VS Code version: Code 1.40.2 (f359dd69833dd8800b54d458f6d37ab7c78df520, 2019-11-25T14:54:45.096Z)
OS version: Windows_NT x64 10.0.18362

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz (4 x 2494)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: unavailable_off<br>skia_renderer: disabled_off<br>surface_control: disabled_off<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off<br>webgl: enabled<br>webgl2: unavailable_off|
|Load (avg)|undefined|
|Memory (System)|3.88GB (0.97GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

Swifster, 2020-01-10:
Got the same issue with my Adafruit Feather nRF52 Bluefruit

Auto-detected: /dev/ttyUSB0
Uploading .pio/build/adafruit_feather_nrf52832/firmware.hex
Traceback (most recent call last):
  File "/home/pi/.local/bin/adafruit-nrfutil", line 3, in <module>
    import re
  File "/usr/lib/python3.7/re.py", line 143, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
*** [upload] Error 1

Swifster, 2020-01-13:
Ok I found something on this problem. I have not tested this but I will give this a try.

Here is the link: https://bbs.sipeed.com/t/topic/1467

ivankravets, 2020-03-01:
Fixed in the dev-platform. Please run `pio update`.

closed 2020-03-01 by ivankravets