platformio/platformio-vscode-ide #1415: Error starting PIO Home

jmshipe, 2019-12-02:
%23 Description of problem
  Fresh install of VSCode 1.40.2 on MacOS Sierra 10.12.6. Installed PlatformIO IDE from extensions marketplace.

  %23 Configuration

  VSCode: 1.40.2
  PIO IDE: v1.10.0
  System: Darwin, 16.7.0, x64

  %23 Exception
  ```
  Error: Error: Detected unknown package 'contrib-pysite'

	at /Volumes/Data/Users/jshipe/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:11965
	at ChildProcess.d (/Volumes/Data/Users/jshipe/.vscode/extensions/platformio.platformio-ide-1.10.0/node_modules/platformio-node-helpers/dist/index.js:1:8215)
	at ChildProcess.emit (events.js:200:13)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
  ```

jhoughjr, 2019-12-08:
I've noticed for a few days that opening home would crash VS.code until I rebooted. Slept 6 hours and started to work again, only to have no status bar buttons and all commands not found. Already tried the fix steps from this issue from 2018. On macOS Catalina latest vs code Home 3.01 Platform 4.1.0

jmshipe, 2019-12-09:
rebooting hasn't fixed anything. uninstalling, deleting, and reinstalling hasnt fixed it, either. If I had to guess, I'd say this issue has something to do with the fact that i'm running on Sierra.

midlll, 2020-02-26:
I have the same problem and I'm using Windows 10 and VSCode 1.42.1... 
@jmshipe have you solved it maybe?

closed 2020-06-03 by ivankravets