platformio/platformio-vscode-ide #1416: Windows 10, VSCode: PIO Home doesn't load

tms320, 2019-12-02:
Hello!
I have installed PlatformIO into my VSCode and two days everything worked fine.
But today "PIO Home" page suddenly stopped to load. When I press PIO Home button the "PIO Home" tab opened but there is only "Loading..." text. And nothing happens. No error messages.
If I run "pio home" in PlatformIO terminal the PIO Home started successfully with the following message:
Open PIO Home in your browser by this URL => http://127.0.0.1:8008
PIO Home has been started. Press Ctrl+C to shutdown.
And then the http://127.0.0.1:8008 page automatically opens in my browser. But something goes wrong:
[image]
I see only rotating dots. No core version, no news. Just "Loading...".
If I press "Libraries" I don't see any libraries. Just "Loading...".
PS: My OS: Windows 10 x64 Home. I had not install any new software or updates in my system.

tms320, 2019-12-02:
Oops. I restarted Windows and the problem disappeared. Sorry to bother you )

Saddamko, 2019-12-03:
Yes, restart Win10 is helped, but it is not good behaviorю

closed 2019-12-04 by tms320